balena-io/etcher #44: Check for SDCard integrity

jviotti, 2015-11-23:


jviotti, 2015-11-24:
> This software will write to all available free blocks on a volume and will then verify (read back) whether the files are still in the right shape. It does this byte-by-byte, so not a single bit will go untested (well, actually, the very last Megabyte might go untested, see the Readme).

http://www.broes.nl/2012/08/verify-the-integrity-of-a-flash-sd-card-on-a-mac/

jviotti, 2016-07-18:
The scope of this project is to implement a routine to detect faulty removable drives (mainly SDCards) early on, which can be implemented as a new NPM module. I didn’t research this in detail, but looks like h2testw/f3 algorithm is the standard way to perform this integrity check. This page contains detailed information about such algorithm: http://oss.digirati.com.br/f3/

The module should provide a function that given a device path (and given the right amount of privileges), returns whether the drive is faulty or not, and does so in a cross platform way.

Given that F3 is open source and already cross platform, we could create a native NodeJS addon around that code, although it’d be more desirable to re-implement in JavaScript unless the complexity is high.

ThomasKaiser, 2016-09-21:
f3 works reliably to test through flash media but can take ages. To detect counterfeit cards faster we use the following method: On media with 16 GB size or above, create partitions 8 GB in size and just test 1MB on each starting from the last partition. Easy to spot counterfeit cards and report real capacity in less time.

jviotti, 2016-09-21:
Hi @ThomasKaiser ,

Thank you very much for the very valuable input on this issue. I'll definitely give the 8 GB partition idea a try!

jviotti, 2018-10-15:
Duplicate of https://github.com/resin-io/etcher/issues/1308

closed 2018-10-15 by jviotti