balena-io/etcher #54: Allow passing a URL and streaming an image to the drive.

jviotti, 2015-11-30:


jviotti, 2016-07-18:
In Etcher, we want to allow the user to pass a URL to an image hosted on the internet, and stream directly to the device. The scope of this project is to implement a new function in etcher-image-stream called “.getFromURL()” that sends an HTTP request to the provided endpoint and makes use of the provided response headers to determine the image size, format, compression, etc to return all the needed information.

If the image is compressed, the function should return the appropriate transform stream that can deal with it (see examples on the same modules).

This functionality should be tested by an end to end suite that starts a local HTTP server to serve example images, and uses those endpoints to write the image to a temporary location, and assert that the image was written and decompressed correctly (if needed).

alexandrosm, 2016-12-02:
For v1 we are ok with two stage download and write, right?

jviotti, 2016-12-02:
> For v1 we are ok with two stage download and write, right?

Yeah, we'll do first download, then write, and we'll improve in the
future.

konmouz, 2016-12-29:
Following #975 concept, would it be possible to have a widget on publisher's webpage that will preload Etcher with a specific image? For instance, on Raspberry Pi's Rasbian page, there could be 3 CTA's: `Download Torrent` `Download Zip` `Flash with Etcher` or `Load to Etcher`.

[image]
 image

alexandrosm, 2016-12-29:
i guess a widget with a dual call to action -> "download including Etcher"
and "open with etcher" (better names needed) would be the ultimate nirvana
for several of our publisher integration efforts. I think there are several
background pieces needed to be aligned, as there are binary signing issues
to sort out, but I love the idea and want to figure out what's needed to
get there.

konmouz, 2016-12-29:
Exactly, pretty much the same way torrents and magnets work.

jviotti, 2017-03-14:
We should re-use https://github.com/zenflow/node-fast-download for this task:

> node module and command-line program for accelerated (multiple connections) http download Stream

lurch, 2017-04-03:
Given the current plans for v2, should this issue be closed?

jviotti, 2017-04-04:
Yeah, lets do it. Given the complexity of streaming *any* kind of images, we decided to only support streaming on our internal image catalog.

closed 2017-04-04 by jviotti