balena-io/etcher #68: Show the Resin.io logo on the footer

jviotti, 2016-01-01:


taahirisaacs, 2016-01-18:
This way should work @jviotti 

[image]

jviotti, 2016-01-18:
@taahirisaacs Looks great! Can you share with me an SVG of the Resin.io logo + text? Also, one observation:

- The footer text currently says "An open source project by Resin.io", but the while thing looks to texty this way. What do you think? cc @alexandrosm

jviotti, 2016-01-18:
It probably will look decent enough. The logo colour and bigger font should balance the text:

[image]

taahirisaacs, 2016-01-18:
@jviotti here you go
[logo-right.svg.zip](https://github.com/resin-io/resin-etcher/files/94460/logo-right.svg.zip)

closed 2016-01-18 by jviotti