balena-io/etcher #74: Add a "burn another" button on completion

jviotti, 2016-01-05:
The completion screen will then have two buttons:

1. Completely start over by reseting every selection.

2. Burning the same image you selected before to another drive (basically keeps the image selection and resets the rest).

taahirisaacs, 2016-01-18:
This method should work @jviotti 

[image]

jviotti, 2016-01-18:
@fedealconada Can you implement the above design? (only the separation between the buttons and explanation. The new footer design is coming in another PR).

jviotti, 2016-01-27:
Re-opening as a reminder to finish the design side of this feature.

commit 6331ee4a (2016-01-20): Add button to burn the same image again. Fixes #74
closed 2016-02-05 by jviotti