balena-io/etcher #127: Could not call remote function ``

jviotti, 2016-01-24:
According to TrackJS, this error happened on Windows 8.1 when clicking "Burn" when a burn was already in process (around ~30%).

```sh
Could not call remote function ``. Check that the function signature is correct. Underlying error: Error processing argument -1.
Error: Could not call remote function ``. Check that the function signature is correct. Underlying error: Error processing argument -1.
    at callFunction (C:\Users\Alexandros\resin-etcher\node_modules\electron-prebuilt\dist\resources\atom.asar\browser\lib\rpc-server.js:203:11)
    at EventEmitter.<anonymous> (C:\Users\Alexandros\resin-etcher\node_modules\electron-prebuilt\dist\resources\atom.asar\browser\lib\rpc-server.js:293:12)
    at emitMany (events.js:108:13)
    at EventEmitter.emit (events.js:182:7)
    at EventEmitter.<anonymous> (C:\Users\Alexandros\resin-etcher\node_modules\electron-prebuilt\dist\resources\atom.asar\browser\api\lib\web-contents.js:129:25)
    at emitTwo (events.js:87:13)
    at EventEmitter.emit (events.js:172:7)
```

jviotti, 2016-01-24:
I can't reproduce in OS X, so it looks like a Windows specific one.

jviotti, 2016-01-26:
Our assumption seems to be incorrect. The burn button is completely un-clickable once the burn starts. The fact that the error mentions a "remote function" makes me think it was a one time IPC error on Electron itself.

I'll keep the issue around for some more time before closing it, in case it happens again.

jviotti, 2016-02-04:
Closing since it doesn't seems to be happening anymore.

Leshich, 2016-06-19:
I got this error trying to make a bootable usb out of Windows_NTFS

```
Sun Jun 19 2016 21:11:13 GMT+0500 (YEKT) Flash ({"image":"/Users/agapov/Downloads/Windows.10/Windows 10.iso","device":"/dev/disk2"})
Sun Jun 19 2016 21:11:13 GMT+0500 (YEKT) Forking: /Applications/Etcher.app/Contents/Resources/app.asar /Users/agapov/Downloads/Windows.10/Windows 10.iso --drive /dev/disk2 --ipc --unmount --check
Sun Jun 19 2016 21:11:15 GMT+0500 (YEKT) Flash error
Error: Could not call remote function ''. Check that the function signature is correct. Underlying error: Error processing argument at index 0, conversion failure from 
Error: Could not call remote function ''. Check that the function signature is correct. Underlying error: Error processing argument at index 0, conversion failure from 
    at callFunction (/Applications/Etcher.app/Contents/Resources/electron.asar/browser/rpc-server.js:218:11)
    at EventEmitter.<anonymous> (/Applications/Etcher.app/Contents/Resources/electron.asar/browser/rpc-server.js:299:5)
    at emitMany (events.js:127:13)
    at EventEmitter.emit (events.js:201:7)
    at EventEmitter.<anonymous> (/Applications/Etcher.app/Contents/Resources/electron.asar/browser/api/web-contents.js:139:25)
    at emitTwo (events.js:106:13)
    at EventEmitter.emit (events.js:191:7)
    at metaToValue (/Applications/Etcher.app/Contents/Resources/electron.asar/renderer/api/remote.js:176:13)
    at Object.remoteMemberFunction (/Applications/Etcher.app/Contents/Resources/electron.asar/renderer/api/remote.js:110:18)
    at showError (/Applications/Etcher.app/Contents/Resources/app.asar/lib/gui/os/dialog/services/dialog.js:97:28)
    at file:///Applications/Etcher.app/Contents/Resources/app.asar/lib/gui/app.js:300:23
    at processQueue (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/angular/angular.js:16170:28)
    at /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/angular/angular.js:16186:27
    at Scope.$eval (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/angular/angular.js:17444:28)
    at Scope.$digest (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/angular/angular.js:17257:31)
    at /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/angular/angular.js:17483:26
    at completeOutstandingRequest (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/angular/angular.js:5955:10)
    at /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/angular/angular.js:6234:7
    at d.util.hasFunction.f.(anonymous function) (file:///Applications/Etcher.app/Contents/Resources/app.asar/node_modules/trackjs/tracker.js:3:492) undefined
 
```

jviotti, 2016-06-21:
Hi @Leshich ,

Thanks for reporting this.

> In conclusion, it would be nice to have some kind of alert or info message, that NTFS USB won't work and I have to make it FAT32 myself (using Disk Utility).

I don't think this is the root cause of the problem. Etcher doesn't care about the drive format since it access the raw disk and blindly writes bytes to it, so it must be something else.

In any case, I've tried flashing to an NTFS drive but as I expected I was not able to reproduce the issue.

Can you actively reproduce when formatting as NTFS? 

What Etcher version are you running BTW?

jviotti, 2016-06-21:
So after some investigation, turns out Electron throws this error when you call a native function of the main process from the renderer process with the wrong arguments.

According to your stack trace, you get an error during the flash process the gets passed to `OSDialogService.showError()` from `lib/gui/app.js`:

```js
// See https://github.com/resin-io/etcher/blob/v1.0.0-beta.8/lib/gui/app.js#L291
.catch(function(error) {

      if (error.type === 'check') {
        AnalyticsService.logEvent('Validation error');
      } else {
        AnalyticsService.logEvent('Flash error');
      }

      self.writer.resetState();
      OSDialogService.showError(error);
    })
```

`OSDialogService.showError()` calls `electron.remote.dialog.showErrorBox()`, a native function from the main process. According to the error, there is a conversion issue with the first parameter, the error title, which is computed like this:

```js
const title = error.message || error.code || 'An error ocurred';
```

The conversion hint makes me thing that either `error.message` or `error.code` (most probably this one) is a number, and not a string.

This issue is obscuring the real bug. I'll submit a PR to make sure `title` is a string as a way to prevent these from happening in the future, but I'd be amazing if you can manage to reproduce the error, so we can hopefully get to the bottom it.

Please let me know if you do have a way to reliably trigger this error and I'll send a custom build to you containing extra debugging information.

Thank you very much again and let me know how it goes!

jviotti, 2016-06-28:
This error has been fixed in https://github.com/resin-io/etcher/pull/497. This of course doesn't solve the root problem you're facing, but `Could not call remote function`, will disappear, giving you the real error you're facing instead, which should be reported as a separate issue.

closed 2016-06-28 by jviotti