balena-io/etcher #129: Blank screen on start

imrehg, 2016-01-25:
Using ArchLinux x64 with `node 5.5.0` and `bower 1.7.2`, upon start I have this blank screen:

[image]

No log in the console beyond the normal node start info. Same thing when creating stand-alone executable with the included Makefile.

`npm test` passes successfully, and `npm install` has only these warnings for optional dependencies.

```
npm WARN optional Skipping failed optional dependency //elevator:
npm WARN notsup Not compatible with your operating system or architecture: elevator@1.0.0
npm WARN optional Skipping failed optional dependency //removedrive:
npm WARN notsup Not compatible with your operating system or architecture: removedrive@1.1.0
npm WARN optional Skipping failed optional dependency /resin-image-write/diskpart:
npm WARN notsup Not compatible with your operating system or architecture: diskpart@1.0.1
```

Don't seem like any already existing issue reported. Any suggestions what to test?

jviotti, 2016-01-26:
@imrehg There are known issues on Electron >0.31.2 on GNU/Linux that prevent Etcher to run correctly on that platform. See https://github.com/resin-io/etcher/issues/52 and https://github.com/atom/electron/issues/3666.

You can open DevTools by pressing `Ctrl+Alt+I`. Can you confirm you see something like this?:

```sh
Uncaught Error: Cannot find module '/home/jviotti/Downloads/7193eec573c62867151d-4e63199c2cf83a13e4a2301c61deec2616f9187b/node_modules/electron-prebuilt/dist/resources/atom.asar/renderer/lib/init.js'
```

We're currently trying to trace this issue in Electron/Chromium but sadly can't give an estimate!

imrehg, 2016-01-27:
Yup, that's the error. Sure thing, I've seen #52, just wasn't sure if this is a duplicate. I can see this is a long-running issue.

jviotti, 2016-01-28:
@imrehg We've been working in this issue and discovered that adding execution permissions for everybody to the home directory fixes the issue:

```sh
$ sudo chmod a+x /home/<user>
```

Can you confirm this works for you as well?

See https://github.com/resin-io/etcher/pull/137

imrehg, 2016-01-29:
Hey @jviotti (thanks a lot for the amount of debugging you are putting into this across repos)

I'm not a big fan of changing permissions on one's home directory. In my experience that usually just fixes the symptoms (not the core issue) and causes side-effects. It should be something wonky about the way the software handles permissions if they need to change outside of their own directories.

Having said that, I've tried cloning to `/tmp/` and working there, and then I can run the program fine (as a workaround).

jviotti, 2016-01-29:
@imrehg Yeah indeed, we're aware of the security concerns. We're just promoting this as a workaround for GNU/Linux while we still investigate the core issue!

jviotti, 2016-01-29:
I added a GNU/Linux section to the README (https://github.com/resin-io/etcher#gnulinux) explaining this issue, and promoting installation on `/opt`, to reduce the security concerns of adding execution permissions to the home directory.

We'll hopefully get to the bottom of it soon!

jviotti, 2016-02-04:
Closing as a duplicate of https://github.com/resin-io/etcher/issues/52. Progress on this will be reported on the linked issue!

closed 2016-02-04 by jviotti