balena-io/etcher #170: Add dfu support

jviotti, 2016-03-01:


lurch, 2016-09-29:
I wonder if this would be stretching Etcher too far outside of its "core mission"?

alexandrosm, 2016-09-29:
Not at all! This is actually part of our to-do list as we add more "IoT
device" functionality. If you're interested in helping get Etcher there,
I'm sure @jviotti would be more than happy to point you in the right
direction.

jviotti, 2016-09-29:
Consider that the main focus of Etcher is flashing IoT devices. If you
look at it from that perspective, dfu makes a lot of sense.

lurch, 2016-09-29:
Ah, fair enough. I was coming from the Raspberry Pi world, where Etcher was being pitched as a cross-platform image-writing utility :-)

jviotti, 2017-04-14:
Check this awesome WebUSB dfu example: https://github.com/devanlai/webdfu/

dragon788, 2017-04-15:
Dfu support would also potentially open up the world of custom keyboards based on the Teensy or other AVR chips? I believe we can use DFU utils to flash the ErgoDox EZ and/or the ErgoDox Infinity if not using the Teensy application.

jviotti, 2022-08-15:
Closing stale issues

closed 2022-08-15 by jviotti