balena-io/etcher #179: Register *.img and *.iso extension with Etcher

jviotti, 2016-03-03:
So the user can double-click on an img/iso and have it opened automatically on Etcher.

hayekr1, 2016-09-30:
I found that this is already implemented

jviotti, 2016-09-30:
What do you mean? In what platform?

jviotti, 2016-12-09:
Turns out this can be done from within the NSIS installer: http://nsis.sourceforge.net/File_Association

lurch, 2016-12-09:
If we do add this, IMHO we ought to prompt the user during installation, rather than doing it automatically.

WasabiFan, 2016-12-09:
I would prefer for Etcher _not_ to do this (at least, not by default). In Windows 10, the default action on opening an ISO is to mount the primary partition and open the mountpoint; I use this regularly for a variety of reasons, including running installers which are distributed as disk images. If Etcher were to take over the default behavior, it would not only make it harder for me to mount disk images (I have no idea how to do this otherwise) but it would break setup guides such as Microsoft's Windows 10 IoT getting started guide, which depends on this action. If an unsuspecting user installed Etcher without realizing the implications of associating the file type, they'd be really confused down the road.

alexandrosm, 2016-12-10:
@wasabifan any similar objections to non-iso extensions?

WasabiFan, 2016-12-10:
@alexandrosm I believe this applies to `.img`s as well. ZIPs and their ilk are also obviously out, given that they are generic file types not specific to Etcher's behavior.

lurch, 2016-12-10:
I guess this argument is effectively a +1 for Etcher creating it's own `.etch` extension, which as mentioned elsewhere I'm not that keen on... (unless it's a metadata-only file, with no .img file inside, in which case `.etch` does make sense)

WasabiFan, 2016-12-10:
Flashing a disk image is an intentional and relatively rare action, so I think it makes more sense for the user to run Etcher and then select a file than the other way around. I don't think that this necessitates a new extension/format nor is it a problem over all.

alexandrosm, 2016-12-11:
It is an argument in favour of .etch for the people that do want that sort
of thing, but not a killer argument.

What I think it does advocate for, however, is some thinking around a
etcher:// protocol that publishers can use in their html to cause Etcher to
come up and handle a specific download/write/provision process. Etcher
could register itself as handling that at the time of install pretty much
unopposed. If our motivation is to save the user having to find/click on
Etcher, this would be another way to go about it. The protocol name should
probably be something other than "etcher" of course.

lurch, 2016-12-11:
Hmm, custom URL-protocol handler is an interesting idea :) Only concern is what happens for users that don't have Etcher installed (yet).

alexandrosm, 2016-12-11:
yeah, it needs to be paired with a "install etcher" process, or some other
wrapping to prevent that :)

jviotti, 2016-12-11:
Nice feedback here! I'll update the issue title to point to `.etch` specifically.

MrDiamond123, 2016-12-27:
I have a idea for .etch files, maybe have them store metadata to download a iso for linux distros on the fly and have a title, author (of iso/linux distro), and a short summary about the iso (or linux distro).

jviotti, 2017-01-23:
Hi @MrDiamond123 ,

We've been drafting a spec of the archive that includes information such as the one you proposed. Feel free to collaborate at https://github.com/resin-io/etcher/pull/707!

jviotti, 2017-01-24:
I'm closing this given the new approach we're going to take towards image metadata.

closed 2017-01-24 by jviotti