balena-io/etcher #202: Scroll or show past 5 mounted devices

danielmahon, 2016-03-10:
It's difficult to select the 6th device mounted, which happens to be the one I usually need.

Window should scroll or extend past frame border (not sure if electron allows that)

[image]

jviotti, 2016-03-13:
Hi @danielmahon 

Good catch. I never tested Etcher with so many drives. I don't think Electron will allow the dropdown to go outside the app frame. I can think of the following possibilities:

- Make use of a native HTML "select" element, which is rendered by the OS and thus will automatically go beyond the app frame if necessary, although might not looks very good (hopefully we can style it enough).

- Make the current dropdown scroll as you suggested, but sounds like a non-optimal user experience.

***

As a side note, I see that some dropdown items also overflow horizontally. Possible solutions:

- Make the dropdown extend vertically if necessary (sounds ugly).
- Increase the width of the application so each step has more space (only mitigates the issue a bit).

I'll ping our designed for input on both.

jviotti, 2016-03-13:
@danielmahon 

By the way, `/dev/disk1` usually represents the system drive in OS X. Is this the case? Etcher should avoid displaying the system drives and only display removable drives. If so, this is a bug.

taahirisaacs, 2016-03-13:
@jviotti how much can we style the native HTML "select" element. Might be the best possible solution to cover all use cases without having to change anything on an edge case basis.

jviotti, 2016-03-13:
@taahirisaacs Agree. I'll give it a try and will post the results here :)

jviotti, 2016-03-13:
We can style the element that triggers the dropdown on click (the `<select>` tag) but not the actual dropdown elements (the `<option>` tags), so I guess this is not an option.

jviotti, 2016-03-15:
@danielmahon Just as an update, we're working on this issue, and discussing about moving the dropdown select drive as a modal, which would solve your issues and align better to some improvements we want to do in the near future.

This is what we have at the moment:

[image]

danielmahon, 2016-03-15:
@jviotti I like it! By the way, in my case, disk2 is the OS. I have a dual boot setup with Mac and Windows (disk1). Other disks are external and internal.

danielmahon, 2016-03-15:
@jviotti I like the idea of showing names instead of physical identifiers, however, for the case of "same name" drives, maybe show path on hover tooltip or something?

jviotti, 2016-03-15:
@danielmahon 

> By the way, in my case, disk2 is the OS. I have a dual boot setup with Mac and Windows (disk1). Other disks are external and internal.

Interesting. We detect the available drives with an NPM modules called `drivelist` (https://github.com/resin-io-modules/drivelist) that we maintain as well. You'll notice there is a `scripts/` directory containing a `darwin.sh` script (https://github.com/resin-io-modules/drivelist/blob/master/scripts/darwin.sh). Could you run that in your machine and share the output here? That'd help a lot!

danielmahon, 2016-03-15:
@jviotti Here ya go. FYI, this machine is a "Hackintosh" so not sure how it determines system disk. In the following `/dev/disk2` is the OSX OS. `/dev/disk1` is the Windows 10 boot drive. Everything else is storage (includeding `/dev/disk0/` which says `system: True`) and formatted FAT, NTFS, or Mac OS Extended (Journaled) for Time Machine.

````
device: /dev/disk0
description: ST31000528ASQ Media
size: 1.0 TB 
mountpoint: 
name: /dev/disk0
system: True

device: /dev/disk1
description: Samsung SSD 850 PRO 256GB Media
size: 256.1 GB 
mountpoint: 
name: /dev/disk1
system: False

device: /dev/disk2
description: Samsung SSD 850 PRO 512GB Media
size: 512.1 GB 
mountpoint: 
name: /dev/disk2
system: False

device: /dev/disk3
description: ST2000DM001-1CH164 Media
size: 2.0 TB 
mountpoint: 
name: /dev/disk3
system: False

device: /dev/disk4
description: Seagate GoFlex Desk Mac Media
size: 2.0 TB 
mountpoint: 
name: /dev/disk4
system: False

device: /dev/disk5
description: hp v290w Media
size: 15.6 GB 
mountpoint: 
name: /dev/disk5
system: False
````

jviotti, 2016-03-15:
@danielmahon Thanks a lot for this. This hasn't being tested in a Hackintosh so I'd bet there are some rough edges there. As far as I can see, none of your drives report a mount point. What OS X version are you running? Also, can you run `diskutil info /dev/diskN` for a couple of them (any will do, maybe a system and a non system one?), just to see the output?

jviotti, 2016-03-22:
Final design:
[image]

I'll start work on this very soon.

danielmahon, 2016-03-23:
Looks great!

---
Daniel Mahon

closed 2016-03-31 by jviotti

jviotti, 2016-04-04:
@danielmahon I've implemented this widget on `master`, and it will be included in the next version. Thanks a lot for reporting!