balena-io/etcher #210: Support burning Windows images

jviotti, 2016-03-15:
Looks like burning Windows images requires a few custom steps for the drive to be bootable:

http://onetransistor.blogspot.com/2014/09/make-bootable-windows-usb-from-ubuntu.html

It'd be nice to experiment detecting a Windows image and perform this steps automatically.

Timopheym, 2016-08-01:
May be for the first step give user a checkbox if he needs those steps? There quite different windows configurations might be.

jviotti, 2016-08-02:
Yeah, sounds like a good way to go. I'll hopefully be looking at this really soon.

Forage, 2016-09-27:
You might want to check out what the official [Windows USB/DVD Download Tool](https://wudt.codeplex.com/) provided by Microsoft does to the USB drive as well. I've been using this tool for Windows 7 without issues whenever I'm being forced to install that OS again.

BioQwer, 2016-12-19:
@Forage i tryed to burn Windows 10 image to flash on MacOS, but it didn't work.
I'm not first time meet this task.
If you resolve this task you get many users on MacOS.
Now there is no simple method for do it.

lurch, 2016-12-19:
See also https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable
And https://gitter.im/resin-io/etcher?at=585358a7c5a4e0233b8ec470 suggests that you can use Bootcamp on MacOS.

BioQwer, 2016-12-19:
@lurch Bootcamp doesn't work for me.
Before using this use should fix configs of Bootcamp. 
And after all, working is not guarantee.

lurch, 2017-02-23:
Somemebody on our gitter channel recently suggested http://www.windowscentral.com/how-create-windows-10-installer-usb-drive-mac for Mac users.

jviotti, 2017-02-23:
That's not entirely accurate, given that in newest MacBook Pro models (like mine), there is no need to create an external bootable drive (you select the image and the Mac reboots straight into it), and therefore you don't even get the change to select an external drive.

See https://support.apple.com/kb/PH25913?locale=en_US

> Depending on the model of your Mac, you might not need an external USB drive to install Windows (you’re asked for one when you prepare your Mac for Windows if it’s required).

jhermsmeier, 2017-02-23:
The BootCamp assistant will add Apple software to the Windows image though.
Also, in the case of creating an installer medium; on macOS Sierra there's no option to flash it to SD cards anymore (only USB thumb drives still work afaik).

Forage, 2017-06-14:
For some more inspiration: https://github.com/slacka/WoeUSB

Lin-Buo-Ren, 2017-09-25:
Hello I'm one of the WoeUSB contributor mentioned above, and I'd like to ask to add WoeUSB as one of the alternative tool that support burning Windows image, refer #1742 for the PR.

zabbarob, 2017-09-29:
Bootcamp didn't provide an option to create a windows USB stick on my late 2009 13" MacBook running OS X 10.11.6 (El Capitan).

However, I was able to boot my PC with UEFI BIOS from the USB stick by simply copying the contents of the Windows 10 ISO to the stick, after erasing it in Disk Utility using MS-DOS FAT and a Master Boot Record as described in this video - https://youtu.be/49NxbzhgsTE

martinbogo, 2017-12-13:
High Sierra has removed the ability to create a bootable USB Windows image.   If we can merge the capability of WoeUSB into etcher, that would solve this headache for a LOT of users.

jhermsmeier, 2017-12-13:
@martinbogo that's odd, I am on High Sierra, and can still do that through the Bootcamp Assistant. Is that missing for you?

martinbogo, 2017-12-13:
Jonas, have you tried since the last update?  I just tried via Bootcamp
Assistant, and there's no option to burn it to USB.   Screenshot attached:

[image: Screen Shot 2017-12-13 at 12.05.01 PM.png]

jhermsmeier, 2017-12-13:
@martinbogo the screenshot didn't make it (GitHub doesn't add them when sent via email); Yeah, updated last night, and it still has that option for me as far as I can see. Did you upgrade from Sierra, or are you on a clean install of High Sierra?

Lin-Buo-Ren, 2017-12-13:
Well just FYI, all WoeUSB magic lies in <https://github.com/slacka/WoeUSB/blob/master/src/woeusb> , the problem is it depend on Linux-only utilities and must be replaced by portable ones.

now-im, 2017-12-25:
Deepin boot maker has been able to flash windows ISO for quite some time.  
You can look into that.

PeterMX, 2018-02-11:
Still waiting for this.

HarshInvader, 2018-04-10:
WoeUSB seems to work for linux. It would be much easier and convenient if Etcher had this feature as well.

Still waiting in
10 April 2018

mrPsycho, 2018-04-21:
yep. still waiting.

zhigang1992, 2018-05-03:
If you're trying to create windows bootable USB, try

http://fgimian.github.io/blog/2016/03/12/installing-windows-10-on-a-mac-without-bootcamp/

Work for me.

Tamkaser, 2018-09-06:
I found a similar  tutorial from Google .

DejavuDmitry, 2018-11-20:
Waiting for...

kriekprinsloo, 2018-11-21:
+1 on this feature

martinbogo, 2018-11-21:
We are getting to the point where we use etcher for most of the work we're
doing at our lab.  Now that we are working with Linux-on-Microsoft and on
ARM devices that support Windows 10, being able to use Etcher to burn
windows images would be a perfect solution allowing us to work with all of
our platforms.  Right now we are using WinTel PC's, Windows on Pi, Linux on
Pi, Linux on PC ( Debian, Ubuntu, et.al. ).

So, add our group to the mix calling for a windows image to USB / Flash /
SD card feature!

Tristan36, 2018-11-22:
UNetbootin could also create bootable USB on MacOS:
https://unetbootin.github.io/
https://www.top-password.com/guide/create-pcunlocker-bootable-usb-from-mac.html

It's great if etcher can also work on MacOS.

ET-Bent, 2019-01-02:
And there we are in 2019. :tada:  
Any updates about Windows image burning? Sure there are other tools that do this but this is the one **essential** feature missing on Etcher to be the one-fits-all tool....

martinbogo, 2019-01-02:
Agreed!  There's absolutely NO legal reason to ignore this.  Being able to
burn windows ISO's into install media is critical.

lurch, 2019-01-03:
...other people claim that #266 is "the one essential feature missing on Etcher" :wink:

wcy940418, 2019-01-05:
Hi all,
I looked for using Etcher to create a flash drive for Win 10 some days before, and found this issue. But after doing some research, it turned out it is pretty easy to make one and might be easy to be integrated into Etcher. Just format the media to FAT32 with MBR and copy all files from Win 10 image to the media, and you will get a bootable Win 10 installation disk. I have verified it. No need to hack anything.

Hope this can help

Reference: https://www.joshbeam.com/2017/11/23/making-a-bootable-windows-10-usb-drive-on-macos-high-sierra/

Lin-Buo-Ren, 2019-01-05:
@wcy940418 Unfortunately the latest Windows 10 installation ISO ships a system image with file size greater than 4GiB, which can't be created in an FAT32 file system.

wcy940418, 2019-01-06:
@Lin-Buo-Ren You are right. I need to reinstall the Win 10 because the ESP was messed up today, and I try to download a new ISO and make the media in the same way, but as you said, the install.wim is larger than 4GB so the FAT32 is not happy with it. But I also figured out another way and it works!

The only thing I need to do in order to make the latest large iso work is chopping the install.wim into small pieces. In windows, DISM could do that, but since I use macOS, I found an alternative (should work on all unix-like system, since the author provide the source code and say so), wimlib, which also does a good job. I split it into two pieces, and each of them is smaller than 4GB. Using them to replace the original install.wim on the thumb drive, and then installation will be all the way same as original one.

The only concern may be the overhead of calling the wimlib to split the .wim file. It will take 1-2 minutes (on my MacBook pro 2016 model).

Testato, 2019-01-13:
+1"for Windows iso support in balena etcher.

windows is the most used operating system on desktoppc, so it is mandatory for etcher support it

OPcake12345, 2019-03-16:
Thanks Tristan36, UNetbootin worked on my early 2009 mac pro (old I know, still relatively fast though) running El Capitan, I have four drives (one with Ubuntu, one with OS X, and one with windows) you helped a lot.

TunnelBlanket, 2019-03-25:
Can't believe they haven't added this yet.

DoogeJ, 2019-05-06:
+1 for this feature, please add this. I was quite surprised and disappointed when I found it it wouldn't work.

cronus00101110, 2019-05-13:
If you add this feature, there'd be no competition in the usb flashing softwares.

dinohorvat, 2019-07-29:
After 3 years seems like still nothing. Please guys

inkdude, 2019-10-15:
+1 windows iso support

nikotidar, 2019-10-25:
+1 windows iso support

StephanMeijer, 2019-11-12:
+1 or windows support

cooldesigns, 2019-12-11:
Awesome app but it is missing a core feature.

OdysLam, 2019-12-26:
+1 for windows support

MyriaCore, 2020-01-08:
HUGE +1 for windows support! Since this issue has been a bit slow coming, if it doesn't seem to be too difficult, it might be wise to implement it ourselves in a community-developed PR.

From a cursory glance, it seems it will take at most two PRs:
1. A PR in this repository editing the [`writeAndValidate`](https://github.com/balena-io/etcher/blob/57145436ab7afef5fb3dc9de9d20fae216d943f7/lib/gui/modules/child-writer.js#L91-L135), in the event that the function call to [`etcher-sdk.multiWrite.pipeSourceToDestinations`](https://github.com/balena-io/etcher/blob/57145436ab7afef5fb3dc9de9d20fae216d943f7/lib/gui/modules/child-writer.js#L115-L121) in the `writeAndValidate` function must be changed to account for changes in the second PR.
2. A PR in the [etcher-sdk](https://github.com/balena-io-modules/etcher-sdk) repository that changes the functionality of `etcher-sdk.multiWrite.pipeSourceToDestinations` to include windows-iso detection and special-preparation functionality.

jwmann, 2020-01-09:
+1 for Windows Support

I've looked into the suggestions of the people in the past and at least for macOS they don't seem to be suited directly for Windows USBs, at least not Windows 10 USBs

UNetBootin doesn't do anything specific for Windows so it has no guarantee of working

As others have mentioned Boot Camp Assistant no longer supports creating a USB installer for anything above High Sierra.
If you go to the support page, there is no mention of an external USB installer, but if you switch to High Sierra, it specifically mentions you can use an external USB.
https://support.apple.com/guide/bootcamp-assistant/install-windows-newer-mac-boot-camp-bcmp173b3bf2/6.1/mac/10.15

polarathene, 2020-01-17:
Landed here from [the Balena Etcher forums](https://forums.balena.io/t/clonezilla-live-usb-fails-to-boot/15341/5). Clonezilla doesn't seem to be able to be detected when flashing the ISO they provide with Etcher to USB, other ISO(Linux distros) I've tried work fine, so I assume it's missing the files(`ldlinux.c32`, `ldlinux.sys`) required like the linked forum thread mentions.

I'll have to find alternative software to get Clonezilla bootable, but it'd be nice if Etcher could add a toggle or auto-detect feature for such.

lurch, 2020-01-17:
@polarathene I suspect a more suitable issue for your Clonezilla request is #413 ?
Or see also https://github.com/balena-io/etcher/search?q=uefi&type=Issues

polarathene, 2020-01-17:
@lurch no, this was the issue that the forum link referred to. So probably the same issue. Other ISO files like linux distro installers flash to the USB fine and can be run, just not Clonezilla, nor Windows ISO(but I haven't tried those with Etcher in a long time)

Clonezilla Live is not all that different to the linux distro installers, except it will clone to a target drive from another instead. No need for persistence.

lurch, 2020-01-17:
I had a bit of a rummage around, and https://clonezilla.org/clonezilla-live.php says
> Clonezilla Live is a small bootable GNU/Linux distribution for x86/amd64 (x86-64) based computers.

and

> Two types of files are available, iso and zip. The former one is for CD, the latter is for USB flash drive.

so it's not surprising that trying to flash the ISO to a USB flash drive doesn't work.
And https://clonezilla.org/liveusb.php talks about manually partitioning the drive and then
> To make your USB flash drive bootable, first change the working dir, e.g. "cd /media/usb/utils/linux", then run "bash makeboot.sh /dev/sdd1"

so unfortunately it sounds like Clonezilla Live isn't distributed in a format that's suitable for flashing by Etcher. :slightly_frowning_face: 

> Other ISO files like linux distro installers flash to the USB fine and can be run

Other Linux distros, like Ubuntu, have solved this problem by using [hybrid ISOs](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos) which means they don't need to supply different versions for CD and USB flash drive (and therefore **do** work with Etcher).

> No need for persistence.

Indeed, that was my misunderstanding of the problem, sorry.

polarathene, 2020-01-19:
@lurch I would say the docs you're referencing are just out of date to be honest. For example, they have a section for [USB with Windows tool Rufus](https://clonezilla.org/liveusb.php#windows-method-e)(an Windows only alternative to Etcher for this task), and it specifically mentions the ISO file:

> Download the Clonezilla Live iso file.
> From MS Windows, run the Rufus program and follow the instructions in the GUI to install Clonezilla Live on your USB flash drive.

And this is exactly what I did. I don't remember the exact message, but Rufus detected some issue with the syslinux setup on the ISO, mentioning two missing or incompatible files, I do recall it saying some version supplied on the ISO was from 10-15+ years ago and Rufus asked to go online and update that with something more recent.

I could perform that again if specifics would help. After Rufus accomplished the task, I booted the Clonezilla Live media just fine and made a backup image of my Windows install. 

Personally, it would be nice if Etcher could support the same feature that Rufus has handled, it apparently does so for other media that needs it(prompt user for permission). I'd rather not need to have access to a Windows system if I could use my preferred cross-platform tool Etcher for this.

lurch, 2020-01-20:
> but Rufus detected some issue with the syslinux setup on the ISO, mentioning two missing or incompatible files

Yeah, that's the kind of logic that Etcher doesn't (yet?) have. That sounds quite different from supporting Windows ISOs, so I wonder if it's worth creating a separate issue? :man_shrugging: 

> After Rufus accomplished the task, I booted the Clonezilla Live media just fine

Nice to know that at least you have a workaround :slightly_smiling_face: 
@thundron Is this something that should be added to the documentation? https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

polarathene, 2020-01-20:
> That sounds quite different from supporting Windows ISOs, so I wonder if it's worth creating a separate issue?

[I created a separate issue](https://github.com/balena-io/etcher/issues/3038). I only chimed in here based on the forum link to this issue being advised as related to the problem. I've not attempted to create a bootable Windows OS for a while, so I can't recall what Rufus does differently vs Etcher to make the ISO bootable if it's different from what happened with Clonezilla.

nicholash777, 2020-01-27:
Support!

f3ndot, 2020-03-07:
I recently went through the hell of running the latest macOS and attempting to "burn" the official Windows 10 iso from Microsoft onto a 16GB USB:

1. The latest iso has an `install.wim` that is larger than the FAT32 4GB file limit, resulting in the file to be missing on the USB (generating quite the annoying error)
2. macOS 10.15 (Catalina) has removed the Bootcamp Assistant approach people have described. So that's out.
3. To side-step the 4GB limit I attempted ExFat as some suggested. However, using ExFat formatting did not make the USB bootable

The **only** way I was able to get a successful Windows 10 USB installation medium with a macOS host was to follow this blog post: https://alexlubbock.com/bootable-windows-usb-on-mac. Spoiler: You have to split the large `install.wim` into 4000 byte chunks (just like rar part files) using some sort of particular splitting binary.

Based on the research I see, Etcher would have to automate the steps of that guide in order to support macOS hosts.

pythonInRelay, 2020-03-11:
How is it after 4 years Etcher still isn't able to create a bootable Windows 10 USB? Guys I love Linux as much as the rest of you, but please get on this asap.

JosephTico, 2020-05-10:
This alone made me uninstall Etcher. It's ridiculous that the developer doesn't care.

thundron, 2020-05-13:
@JosephTico Remember that the "developer" is also you - this is an open source project and we're already working on a number of other things that you may or may not know, but saying we don't care can't be more wrong.
If anyone feels like implementing this, feel free to, it isn't a priority of ours for now

pythonInRelay, 2020-05-13:
Apparently I'm still subscribed to this. Whilst I don't agree with Joseph's attitude :( I really have to reinforce to whomever reads this. Support for burning Windows .iso to USB is of paramount importance and should be the number one priority moving forward. If this application is to be offered at a cost for mass flashing of bootable media, it does not make any sense that Windows is not a supported OS.

I hope this is prioritised in the future, because if Windows was supported I (and I am sure a great deal of other people) would not see remotely any need to use other applications whatsoever.

My best to the Balena team. Have a lovely Summer/Winter everyone.

lurch, 2020-05-13:
> Support for burning Windows .iso to USB is of paramount importance and should be the number one priority moving forward.

That may be of "paramount importance" to you, but that doesn't mean that it's equally important for Balena, which is the company paying for @thundron and @zvin to be working on Etcher.

> If this application is to be offered at a cost

?? AFAIK Etcher has always been (and always will be) Open Source Software, downloadable for free.
If you're talking about EtcherPro, then I think I remember reading somewhere that you'll (obviously) be paying for the hardware, but that the software itself will remain Open Source.

[12 more replies]