balena-io/etcher #212: Wrong path for Downloads and Documents

danielmahon, 2016-03-20:
Not sure if this is unique to my system (hackintosh), but the `Documents` and `Downloads` folder in the sidebar reference the wrong path for OSX.

`/private/var/root/Downloads` should be `~/Downloads` or `/Users/{username}/Downloads`

[image]

jviotti, 2016-03-22:
Hi @danielmahon , sorry for the delay, GitHub didn't notify me of this issue.

I can't reproduce this on any of my Macs, so I guess its unique to your system. I'd suggest opening an issue on the Electron GitHub repository, stating that `dialog.showOpenDialog()` (http://electron.atom.io/docs/v0.37.2/api/dialog/) exhibits this error on a Hackintosh.

jviotti, 2016-04-04:
@danielmahon Any updates on this? Did you create an issue on Electron?

I've been taking a look myself, and Electron uses the standard `NSSavePane` without any electron-specific modification. Do you get this behaviour in other OS X apps apart from Etcher?

If not, this might be related to the fact we perform elevation on Etcher. You could try running another OS X app with `root` privileges and see how it goes.

danielmahon, 2016-04-05:
@jviotti I tried just running `dialog.showOpenDialog()` inside a simple electron app and it works fine, opens the dialog box, shows favorites and the correct paths. Ill try again with elevated privileges and see what happens.

jviotti, 2016-04-05:
@danielmahon Awesome, let me know how it goes. I re-run the experiment here, but everything looks fine.

jviotti, 2016-04-21:
Closing since this happens due to Etcher running as `root` with no easy workarounds. We're experimenting with only elevating a child process to perform the write, which should fix this issue.

closed 2016-04-21 by jviotti