balena-io/etcher #215: OS X: Uncaught exception thrown if no password is given on startup

lekkas, 2016-03-23:
If I click 'Cancel' when the admin password dialog pops up on program startup the following exception is thrown after the 'Elevation Error' message:
[image]

Minor bug really, more UX related since it doesn't affect program usability.

jviotti, 2016-03-25:
@lekkas Thanks for the heads up! Sorry for the late response, turns out I wasn't "watching" the repo and therefore got no notifications :)

I'll send a fix very soon.

jviotti, 2016-03-25:
PR sent, thanks a lot!

lekkas, 2016-03-25:
Cool! By the way, I've been using Etcher since last week and there is no going back when it comes to burning SD cards! :+1:

jviotti, 2016-03-25:
Awesome to hear that! Ping me with any suggestions :)

closed 2016-03-25 by jviotti