balena-io/etcher #250: Add "skip" link below validation progress bar

jviotti, 2016-04-07:
Add am underlined "skip" link in validation stage under the progress bar. Once the "skip" link is pressed, it should then ask: "Skip once?", "Skip always?".

jviotti, 2016-04-12:
[image]

jviotti, 2016-04-12:
[image]

jviotti, 2016-04-12:
As discussed, this can be simply reduced to "Skip".

jviotti, 2016-04-12:
We're pending designs for the modal that appears after the user clicks "Skip", which asks him if he wants to skip just once, or always.

Josephvb10, 2019-04-30:
3 years and still not implemented such a simple feature?

jeffersonest, 2020-04-04:
> 
> 
> 3 years and still not implemented such a simple feature?

4 Years***

whatbh, 2022-05-02:
5 years

jviotti, 2022-08-15:
Closing stale issues

closed 2022-08-15 by jviotti