balena-io/etcher #265: Backup SD Card / Wipe Clean

nmaas87, 2016-04-08:
Could you add an "Wipe Clean" only option to format an SD Card and an "Backup to Image" option, so that the SD Card would be read and saved as compressed Image to the harddrive (maybe gz or 7z) - for Backup/Archive/Clone purposes? If your "Clone to many SD Card Readers at the same time" feature would make it to etcher, than this would really make it the "go to" utility for all use cases :)!

Anyway :+1: Good work and keep it up :)!

jviotti, 2016-04-08:
Hey there @nmaas87 , thanks for the suggestions. To summarise:

- [ ] Add "wipe clean" option

This option would basically erase your removable drive writing zero bytes all the way, right? Alternatively, are you expecting Etcher to format your drive as FAT or something like that?

- [ ] Backup to image

This is a very nice suggestion. Since Etcher already understands ZIP archives, it'd be nice to implement this feature as "save drive contents as an `*.img` file" and compress with ZIP.

- [ ] Clone to many SD Card Readers at the same time

Definitely! This is an amazing feature, and we're working hard on it, so stay tuned. I have a proof of concept, however is still needs a lot of polishing. Currently, it causes a kernel panic on OS X quite often :)

nmaas87, 2016-04-08:
> 
This option would basically erase your removable drive writing zero bytes all the way, right? Alternatively, are you expecting Etcher to format your drive as FAT or something like that?

Formating it to FAT would be cool, so that you can just "erase" SD Cards if their content is not needed anymore. This is a problem, especially on Windows: SD Cards with Partitions (i.e. EXT3,4 etc.) cannot be formated to their "factory state" without using "Higher End Tools" like diskpart or the SD Formatter Tool. With "Windows Only" you can only format the FAT Partition of the SD Card (i.e. RPi config/kernel/boot partition) - however it leaves the ext partition untouched. And that is only half the job done, as some tools can't select the root SD Card Drive to overwrite that, but only see the small formated FAT partition.

> 
This is a very nice suggestion. Since Etcher already understands ZIP archives, it'd be nice to implement this feature as "save drive contents as an *.img file" and compress with ZIP.

That would be very cool and space saving at the same time. Often tools only create img files without any compression - or some binary blob which is compressed in some proprietary format, which can not be opened by other tools. So this solution seems nice.

> 
Definitely! This is an amazing feature, and we're working hard on it, so stay tuned. I have a proof of concept, however is still needs a lot of polishing. Currently, it causes a kernel panic on OS X quite often :)

Thanks, I am using dcfldd at the moment to do this in Ubuntu, but it would be awesome to do this stuff on Windows as well. I often create customized Raspbian Versions which I install on one RPi, do my changes on that system, backup the ready image to my pc and then massclone this image via different tools with my 10 Micro SD Card Reader Clone Station (for use on Raspberry Jams and Workshops and such :)). Thats why I am asking for these features, as this would make life a lot easier and I could do all this work from Ubuntu and Windows then :).

Oh, and one last "Feature Wish" which sounds a bit crazy and should be an "Advanced Feature" with Red Blinken Lights and Warnings and such: It would be helpful if you could turn off the integrity and length check for the images. Reason why is following: You create an RPi Image on an 8 GB SD Card and copy this card to your PC. There is only 2 GB on this SD Card used in the File System, however, the Partition has been enlarged to the full 8 GB. Your "dd-Type" Image will be 8 GB in size (even though zip will make it around 2 GB now with your new feature, as it will "zero-out-all-zeros" ;)) - if you now try to etch that 8 GB Img to an 4 GB SD Card - which will be fine in all ways - your tool would say "No". Why? Because a) Partition Size was 8 GB, and even though it will perfectly work - sanity check will say no. So there should be an option to shut that off, so that you could burn your 2GB-in-8GB-Pelt Image to the new 4GB Home :). (I used dd on Linux and an very old Version of WinDiskImager on Windows to do that - and worked a treat. Sometimes you need that because you are too lazy (or simply out of time) to trim the partition sizes on your local image down and truncate the rest of the image...))

Thanks a lot :)!

jviotti, 2016-04-08:
Hey there,

> Formating it to FAT would be cool, so that you can just "erase" SD Cards if their content is not needed anymore. This is a problem, especially on Windows: SD Cards with Partitions (i.e. EXT3,4 etc.) [...]

I see. My only worry is that formatting to FAT might be a bit too opinionated. It might make sense in the context of SDCards, but Etcher writes any image file to a removable drive, including but limited to SDCards. I guess we could provide a way to format to various well known formats, however that might add unnecessary complexity.

A solution which I think might fit you (let me know what you think) would the following (I didn't try it, but should work):

- We can provide a "Wipe" feature in Etcher that erases the Master Boot Record of a device, causing the drive to lose information about its partitions, and thus appear as "empty". Now that the partitions are cleared, you can format the whole drive with the available tools in your operating system without any issue.

> That would be very cool and space saving at the same time. Often tools only create img files without any compression - or some binary blob which is compressed in some proprietary format, which can not be opened by other tools. So this solution seems nice.

Great, I opened a new GH issue to track and implement this feature: https://github.com/resin-io/etcher/issues/266

> Thanks, I am using dcfldd at the moment to do this in Ubuntu, but it would be awesome to do this stuff on Windows as well. I often create customized Raspbian Versions which I install on one RPi, do [...]

I feel your pain :) Created an issue here: https://github.com/resin-io/etcher/issues/267. I'll be posting updates periodically there.

> Oh, and one last "Feature Wish" which sounds a bit crazy and should be an "Advanced Feature" with Red Blinken Lights and Warnings and such: It would be helpful if you could turn off the integrity [...]

Actually, you can already disable the validation check by clicking the settings cog icon at the top right corner of the window, and unchecking "Enable write validation on success".

Regarding the size mismatch, we take care of this when performing the validation (we only get back the necessary data and not the whole drive for performing the check), and I guess we could take care of this as well in the "Backup" functionality (e.g: only copying back the necessary amount of data), although not sure if that'd be possible.

In any case, disabling validation from settings should be good enough for you!

nmaas87, 2016-04-08:
> 
We can provide a "Wipe" feature in Etcher that erases the Master Boot Record of a device, causing the drive to lose information about its partitions, and thus appear as "empty". Now that the partitions are cleared, you can format the whole drive with the available tools in your operating system without any issue.

That should do it :)!

> 
In any case, disabling validation from settings should be good enough for you!

Thanks a lot for the hint, I will give that a try soon.

jviotti, 2016-04-11:
Okay, awesome. I created a new ticket for the "wipe feature" here: https://github.com/resin-io/etcher/issues/273.

I'll close this one so we can continue discussion in each appropriate ticket.

closed 2016-04-11 by jviotti

adrianmihalko, 2016-10-31:
Guys, Backup to image is still not implemented?

dlech, 2016-10-31:
@adrianmihalko, #266