balena-io/etcher #266: Implement a way to perform a backup from a drive

jviotti, 2016-04-08:
From https://github.com/resin-io/etcher/issues/265

This option would copy whatever there is on the device as a local `*.img` file, and will compress it as a ZIP archive, in a way that can be easily burnt back with Etcher.

jviotti, 2016-12-05:
I just had some ideas for this: The problem at the moment is that blindly copying the device contents back means copying a lot of unnecessary data and having a backup image that equals the drive size. If you want to flash that back to a drive that is smaller, but in which the image logically fits, you'll have a hard time figuring out how to reduce appropriately. 

As a solution, we can scan the partition table and only copy what is really necessary.

lurch, 2016-12-05:
> As a solution, we can scan the partition table and only copy what is really necessary.

I thought I'd already suggested that, but maybe I'd imagined it and it was something I was *going* to suggest, but forgot to :confused: 
Sounds good :+1:

jviotti, 2016-12-05:
Haha, maybe it was on another related issue and it somehow got lost?

alexandrosm, 2016-12-07:
@jviotti, reading this again, I think you're making your life difficult. We can start with a "dumb" imaging process, and improve on that later

lurch, 2016-12-07:
Reading the partition table to determine how much of the disk to _actually_ read, and simply outputting a smaller diskimage at the end of the process, isn't that complicated IMHO :-)  It's only a small part of the process, so I'd see no reason to leave it out.

jviotti, 2016-12-07:
@alexandrosm The thing is that copying back the whole thing could bring a lot of issues in the case of Etcher. Most users get away with it using something like `dd`, since `dd` will only copy what it can and discard the rest (so doing a backup of the whole thing causes no issues most of the time), however Etcher will complain in that case, which means users can only backup drives that they will re-flash to the same drive or a larger one.

Maybe I'm indeed overthinking it and that's not an usual scenario (maybe people only re-write the backups to the exact same drive most of the time), but if it is, Etcher would feel really awkward.

alexandrosm, 2016-12-08:
All I'm saying is, let's do the part that's easy now, and solve the hard
part later, if its an issue, rather than block the easy on the hard :)

agsdot, 2017-01-24:
Any updates/progress on this issue @jviotti?

blinkenlight, 2017-01-24:
This would be pretty important for me too - I'd just be doing backups and restores (if needed = I screw up) on the same card anyway.

jviotti, 2017-01-24:
Hey there,

We are sadly not making progress on this one now given that we're preparing some other major features for a new release. This is something we really want to support, so we'll definitely allocate some time for this soon.

That being said, if anyone is motivated enough to make this work, we welcome contributions to https://github.com/resin-io-modules/etcher-image-write, and then to integrate it with the UI (ping me on [Gitter](https://gitter.im/resin-io/etcher) if you're interested and I can provide guidance to implement this)

archesys, 2017-04-12:
Hi all,

I discover your application some days ago and find it very nice and useful, I work on Linux and a friend on Apple, so I also recommend this application to him.. We work on raspberry so we regularly need to backup and restore sdcard.. The backup function (in zip or tar.gz) would be very useful. no more news on that ? I'm french so if you want me to translate interface in french, I could help you (I don't have so much time but If it's not too big work it may be affordable.

Eric..

fzymek, 2017-05-18:
+1 for this feature from me as well, very useful for raspberry

tomatac, 2017-05-31:
This is an essential feature. Currently, what is the recommended way to create an image?

jviotti, 2017-05-31:
@tomatac You can use `dd` to copy the device data to a file.

DWatkins, 2017-07-16:
+1 

A backup feature would greatly ease my workflow

adrianmihalko, 2017-07-29:
Still no backup feature?

dominikkv, 2017-08-19:
+1
Would be great!

philomatic, 2017-08-30:
Do this!

+1

onkelandy, 2017-08-31:
Alternative way is to use Apple Disk Utility.. But having a simple way to create an image with only the necessary size would be aaaaawwwwesome!
+1

borpin, 2017-11-17:
+1 Would be fantastic to have this and not have to go back to dd 👍

jarrah31, 2017-12-01:
21 months since the original request - I guess this feature isn't going to be implemented?

lurch, 2017-12-01:
@jarrah31 This is still on the TODO list (and in fact #1671 was a big step towards making this backup feature possible), we've just had lots of other more important features to concentrate on.

xristocodos, 2018-01-16:
So even though most people aren't writing the image back to the exact same drive (usb id), they are most certainly moving it to the same SIZE of drive or greater. Just make a disclaimer and fade out the Begin! button (or whatever) if it doesn't detect an adequate size. Cramming images into smaller drives is a dicey tricky artform that I personally don't believe anyone expects Etcher to perform. I'd just put a disclaimer saying that 0.) this feature is somewhat experimental, 1.) the drive must be equal or larger capacity, and 2.) if it is bigger, it will "appear as" the size of the image, until its reformatted at some point. If i need to copy an 8gb uSD stat and all i have is a 16gb usb stick, i'm not really concerned about the extra space, I probably just need the image duped asap.

sunapi386, 2018-01-19:
Do we have an ETA on this?

lurch, 2018-01-19:
This is a feature we'll need for https://etcher.io/pro so it'll probably get added sooner rather than later ;-)

KerstinMaur, 2018-02-28:
[Etcher Pro](https://etcher.io/pro/) and [this feature](https://github.com/resin-io/etcher/issues/266) are awesome! 😍
Looking forward to them! 🧐

mirceal, 2018-06-24:
+1

Any update on this?
Thanks

jhermsmeier, 2018-07-02:
Sure, the technical foundation is pretty much there, and the user interface part is being worked on soon, and this will most likely end up in the next release :)

MDM63, 2018-08-11:
Is this going to be a feature for Etcher or only for Etcher pro?

jhermsmeier, 2018-08-28:
@MDM63 This is going to be available in both Etcher & Etcher Pro; generally both of them will maintain feature parity in regards to the writing process.

cjcharles0, 2018-09-11:
I dont want to appear ungrateful, but is there an ETA on this yet? Such a useful feature (even if it doesnt scan the partition table to avoid copying blank space)

TommySharpNZ, 2018-09-14:
+1 for this feature too please!

kriekprinsloo, 2018-09-15:
+1 I also think that this would be great.

richardhughes260, 2018-11-03:
Is there an ETA on the backup feature?

ddomit, 2018-11-09:
+1 Any news?

dhowe, 2018-12-24:
progress?

JaffaBoy, 2019-01-26:
hows this coming along?

boelle, 2019-02-04:
yep... seems a shame that we can create an image from the sd card :-(

Poincare1, 2019-04-16:
Any progress in being able to create an image from flash media?  Seems it was reported more than a year ago it would be an upcoming feature.

sunapi386, 2019-04-26:
Clonezilla is a capable alternative, if you don't mind rebooting booting from USB and waiting. It's not bad if you dedicate a Raspberry Pi to do it. https://clonezilla.org/show-live-doc-content.php?topic=clonezilla-live/doc/01_Save_disk_image 

Clonezilla source code and see example bash scripts for cloning in "samples" folder. https://gitlab.com/stevenshiau/clonezilla.git

PTS93, 2019-06-25:
@jhermsmeier is this feature put on the backlog for now? It sounded like it was close to release a year ago.

jhermsmeier, 2019-06-25:
@PTS93 I don't know, I'm not working on Etcher anymore, but maybe @thundron or @zvin can answer that question

paddywwoof, 2019-08-12:
Any news? @thundron  @zvin

thundron, 2019-08-12:
@paddywwoof Working on it, but it will still need a bit more before it's pushed and merged, it requires some changes in the UI flows too

rin67630, 2019-08-13:
By the way: to get rid of the useless data, you could implement a simple preliminary step: write a growing file with zeroes to the DS card until until it returns "full", then delete that file...
The compressed result should be perfect...

mitra42, 2020-01-22:
Judging by the length of this topic, I guess this is never going to happen.  RPI's do a good job (mostly) with this with their SD Card copier, but seems to be incompatible with something (maybe usbmount)  - would be great to see this is in Balena Etcher

jadonk, 2020-03-12:
I'd really like to see this pushed across the finish line.

kevincastejon, 2020-09-17:
You should have listened to @alexandrosm 3 years ago when he was proposing to implement a "dumb" copying of the image as a start and improve it later..
I'm really waiting for this feature to come.

rradar, 2020-10-28:
This "feature" is present in [usbimager](https://gitlab.com/bztsrc/usbimager/) btw :rocket:

blinkenlight, 2020-10-28:
@rradar Thank you kind sir, your comment easily being the single most useful one on this ancient thread - which I can now finally leave behind - considering the fine gentlemen in charge of this here repo clearly put continental tectonic plates (and even some high-profile open source bug trackers with ongoing decade-old unresolved major issues) to shame in their eagerness to implement an obviously widely desired dead-simple feature. As to the fine gentlemen themselves: it's okay, you may censor this post as well if you wish; you've read it and that was the point - also, you are aware everyone who asked for this feature already got an email with the stuff you oh-so-promptly removed, right...?

iKarith, 2020-10-31:
I'm no longer looking for this feature in balenaEtcher.

I'd suggest that garnering a reputation for deleting criticism isn't good PR, but doubt anyone at balena cares what I think.

thundron, 2020-11-05:
@iKarith given it wasn't criticism but just banter and veiled insults for a completely different issue (not that would be justified in a different issue), yes, we don't want those to appear anywhere ideally and it's not the first time either.
Now on criticism - it can be hard, but we don't and will never disapprove it and. we even welcome it as much as we can - when it stays in the boundaries of criticism, that is.
Let's keep the discussion on the subject now - this issue is for backing up a drive and I'd like it to stay so :)

shawaj, 2020-12-13:
For anyone just finding this you can now clone devices in balenaEtcher since v1.5.107

You can find the latest versions on the [balenaEtcher website](HTTPS://etcher.io) or from the [GitHub latest release page](https://github.com/balena-io/etcher/releases/latest)

lurch, 2020-12-14:
> For anyone just finding this you can now clone devices in balenaEtcher since v1.5.107

But I guess there's still a slight difference between "clone" (i.e. copy the contents of one SD card to another SD card) and "backup" (i.e. create an image of your SD card on your hard drive) ? :wink:

shawaj, 2020-12-14:
@lurch yep for sure! Have left it open as it is not really a fix - but does allow you to "backup" an image - at least in one sense of the word :-)

daviderickson, 2021-02-22:
Etcher is so good at writing images, it would be great if the other half of the pipeline of creating images could be covered.  And in particular, as the size of cards continues to grow, having an ability to go straight to a compressed image with no intermediate uncompressed file would be tremendously useful.

thundron, 2021-03-01:
@daviderickson that would mean having the image information *outside* of the compressed image, which I'm not sure is possible, or decompressing directly to the media drive which can be significantly slower than just decompressing and *then* flashing

daviderickson, 2021-03-01:
> @daviderickson that would mean having the image information _outside_ of the compressed image, which I'm not sure is possible, or decompressing directly to the media drive which can be significantly slower than just decompressing and _then_ flashing

What image information are you referring to?  .img files are raw byte by byte transfers, there wouldn't be any need to decompress part of it to determine how to write it.  No question that writing a compressed image would likely be slower than an uncompressed one, but if it saved many GB of temporary disk usage and writes in the meantime, I'd gladly make the trade.

charettepa, 2022-03-01:
any update on this, it seems to have been in the works since 2016

doug62, 2022-03-07:
+1 This - The ability to copy a snapshot/image to a host iso so that they can be cloned later is critical (and so easy given the app's current features)

[15 more replies]