balena-io/etcher #267: Implement parallel writes

jviotti, 2016-04-08:
From https://github.com/resin-io/etcher/issues/265

Allow a single image to be burnt to many SDCards at the same time.

lurch, 2016-09-29:
And I guess this will also need to cope with some of the drives being written to having much slower write speeds than others (e.g. a Class4 SD card in a USB2 cardreader vs. a Class10 SD card in a USB3 cardreader). So perhaps there needs to be separate decompression queues for each drive, rather than trying to stream the same decompressed data to each of the drives? Unless of course you're happy for the whole process to just run at the speed of the slowest drive.

jviotti, 2016-09-29:
Hey there,

That's spot on. We've experimented with opening different file
descriptors from the input image, and handling decompression separately,
which helps avoiding faster drives being bottlenecked by slower drives,
however that also means that when uncompressing many multiple streams
(think >20), the CPU usage rises a lot, and we're basically duplicating
computation.

Another thing we found out is that performing much I/O on a single
NodeJS process (even with different file descriptor) slows things down a
lot, given the single-threaded nature of NodeJS. Seems like the way to
go is to spawn multiple NodeJS processes.

lurch, 2016-09-29:
Yeah, and I guess by running all-drives at the speed of the slowest drive, you only need a single progress bar, rather than separate progress bars for each drive ;-)

alexandrosm, 2016-12-07:
@jviotti I think we can push this off the v1 release.

jviotti, 2016-12-07:
Done!

jviotti, 2017-04-25:
Lets try using web workers with node integration for this: https://github.com/electron/electron/pull/8852

jhermsmeier, 2018-03-23:
Resolved by #2124

closed 2018-03-23 by jhermsmeier

unphased, 2018-08-26:
I've flashed about 20 cards now and I have noticed that the time taken more or less scales linearly with the transfer speed of one card. e.g. if I have 3 cards that are independently capable of flashing @21MB/s, flashing 3 of them simultaneously has them proceed at almost exactly 7MB/s, making it so that I have to wait 3 times as long for the batch. Now, this is a process improvement, however I am confused as to why there is no speedup as you would see if you just ran concurrent `dd`s on multiple drives. 

This is using a MacBook with multiple usb-c to sd card adapters.

sonttran, 2020-07-07:
Same issue, same setup with @unphased on **Etcher Version 1.5.100.** Any way to improve speed to full in parrallel, @jviotti ? That would be great.

> I've flashed about 20 cards now and I have noticed that the time taken more or less scales linearly with the transfer speed of one card. e.g. if I have 3 cards that are independently capable of flashing @21MB/s, flashing 3 of them simultaneously has them proceed at almost exactly 7MB/s, making it so that I have to wait 3 times as long for the batch. Now, this is a process improvement and already a big win, however I am confused as to why there is no speedup as you would see if you just ran concurrent `dd`s on multiple drives. The reason I use etcher is so I can avoid building a `dd` runner and figuring out how to properly verify a disk, but this seems... not ideal.
> 
> This is using a MacBook with multiple usb-c to sd card adapters.

socks415, 2025-03-05:
I know this is old, but it's 2025 and I'm on a macbook air m2 with the latest etcher and experiencing the same thing trying to flash multiple cm4s at the same time.  one cm4 flashes at roughly 5.7mb/s. 3 flashes at about 1.9mb/s. it's completely linear.  It's also very unstable and fails to successfully complete often when flashing more than one at a time