balena-io/etcher #271: Running as non-Administrator on Windows 10 Results in Electron No-Write-Permissions Error

xd1936, 2016-04-09:
Hello!

If I run Etcher on Windows 10 x64 as my default user, I get a User Account Control prompt as expected. Even if I select Yes and give Etcher Administrative permissions, once I start burning an image to an SD card, I get the following error in a dialogue box:

    Uncaught Error: EPERM: operation not permitted, write

The progress bar makes it from around 0% to 3% before failing.

This is with Windows 10 Pro x64, a USB 2.0 SD card reader plugged into a USB 3.0 port, a Sandisk MicroSD Card, and the Raspbian Lite img file.

xd1936, 2016-04-09:
Perhaps related to [electron#2033](/electron/electron/issues/2033)?

jviotti, 2016-04-11:
Hi @xd1936 

Thank you for reporting this. We used to get this error when writing directly to a physical drive in Windows, due to policies in the operating system better described in [this article](https://msdn.microsoft.com/en-us/library/aa365747(VS.85).aspx). To summarise:

> If you write directly to a volume that has a mounted file system, you must first obtain exclusive access to the volume. Otherwise, you risk causing data corruption or system instability, because your application's writes may conflict with other changes coming from the file system and leave the contents of the volume in an inconsistent state. To prevent these problems, the following changes have been made in Windows Vista and later:
>
> A write on a volume handle will succeed if the volume does not have a mounted file system, or if one of the following conditions is true:
>
> - The sectors to be written to are boot sectors.
> - The sectors to be written to reside outside of file system space.
> - You have explicitly locked or dismounted the volume by using FSCTL_LOCK_VOLUME or FSCTL_DISMOUNT_VOLUME.
> - The volume has no actual file system. (In other words, it has a RAW file system mounted.)
>
> A write on a disk handle will succeed if one of the following conditions is true:
>
> - The sectors to be written to do not fall within a volume's extents.
> - The sectors to be written to fall within a mounted volume, but you have explicitly locked or dismounted > - the volume by using FSCTL_LOCK_VOLUME or FSCTL_DISMOUNT_VOLUME.
> - The sectors to be written to fall within a volume that has no mounted file system other than RAW.

If we failed to conform to these policies, we got an `EPERM: operation not permitted, write` as the one you mention. At the moment, we do the following:

- Write zero bytes to the first sector of the disk (the MBR, since Windows freely allows writing to boot sectors).
- Force Windows to "detect" the changes, with `dispart rescan`.
- Write the actual image to the drive (since now we're writing to a physical drive that has no file system).

I believe the permissions issue most probably related to this policies than the issue you linked to, but maybe worth checking anyway.

Can you try other images, and potentially other drives (maybe a USB drive rather than an SDCard reader) and report back how it goes? I remember I experienced some very similar issues specially with USB SDCard readers.

jviotti, 2016-04-11:
Hey @xd1936 

Do you get a stack trace for that error on the error dialog? If so, can you paste it here?

jviotti, 2016-05-12:
Closing due to inactivity. Please re-open if you can share more information about the error!

jviotti, 2016-05-12:
Actually, I just noticed this is a duplicate of https://github.com/resin-io/etcher/issues/334, so please continue any conversation there.

closed 2016-05-12 by jviotti