balena-io/etcher #272: Set minimum window size on Linux

danrabbit, 2016-04-10:
It looks like there isn't any minimum window size. You can pretty easily make the app look bad by resizing the window too small

[image]

jviotti, 2016-04-11:
Hi @danrabbit 

Thanks for reporting this. We are setting `resizable: false` in the `BrowserWindow`. After some more investigation, turns out there was a known issue in Electron where the window was resizable on GNU/Linux even when passing `resizable: false`. I upgraded to the Electron version containing the fix.

Thanks you very much again!

danrabbit, 2016-04-11:
Thank you for the very speedy response!

closed 2016-04-11 by jviotti