balena-io/etcher #276: Create deb and rpm packages for GNU/Linux

jviotti, 2016-04-11:


vankasteelj, 2016-04-13:
I've seen 2 ways of doing that, apparently https://www.npmjs.com/package/nobin-debian-installer can do it on any platform, but I couldn't get a good enough build.

I also have developped a [bash script](https://github.com/butterproject/butter-desktop/blob/master/dist/linux/deb-maker.sh) (available on linux only) to automate `fakeroot dpkg-deb` task

jviotti, 2016-04-13:
Hi @vankasteelj 

Great information, very appreciated! What do you mean by:

> but I couldn't get a good enough build.

?

vankasteelj, 2016-04-13:
I don't remember specifics, I just remember trying it, and it wasn't doing what I wanted, either non-working at all, or not implementing scripts like I wanted. Sorry

jviotti, 2016-04-19:
I'm closing this issue given I believe creating `deb` and `rpm` packages for Etcher is not worth it after checking out [AppImage](http://appimage.org). They are "nice to have", but the premise of AppImage to provide a single distributable executable that works across distributions lines up better with Etcher intentions, and provides [a mechanism to perform auto-updates](https://github.com/resin-io/etcher/issues/336) that is more similar to the experience in the other supported operating systems.

Debian and RPM packages might be nice to have in the future, but for now they mean extra maintenance  without a clear benefit.

AppImage support has been implemented in https://github.com/resin-io/etcher/issues/323

closed 2016-04-19 by jviotti