balena-io/etcher #287: Normalise image open dialog when running as root in OS X

jviotti, 2016-04-11:
The open dialog from Etcher differs from other applications (like missing Favourites) given that Etcher is ran as root.

alexandrosm, 2016-04-12:
Could Etcher start a non-root sub-process to select the file?

jviotti, 2016-04-12:
@alexandrosm The problem is that once we package the application, the "Electron" binary is locked down to running a certain application (the one it has been packaged for), so if we want to fork a child process, we can't have Electron run it (e.g: we should make assumptions like NodeJS/Electron already being installed on the system). This same error applies when only asking for elevation on the burn phase.

We could maybe try having the application's entry point react differently to certain command line arguments, but then its also a pain to communicate between the processes correctly, and sounds like a mess (which I'd like to avoid).

PR #423 (2016-06-21): [WIP] Only perform elevation of writer process
This is a big PR that solves a hard issue we've been carrying over for
too long. The Etcher GUI application now runs an intermediary script
called `writer-proxy.js` which elevates itself if needed, spawns the
Etcher CLI piping `stdout` to a temporary file, tails that file and uses
IPC communication to share back the results with the Etcher GUI.

Fixes: #412
Fixes: #351
Fixes: #287
Fixes: #444
closed 2016-06-21 by jviotti