balena-io/etcher #289: Elevation error

loranger, 2016-04-12:
Hello,

On OSX 10.11.4 I cannot launch Etcher : As soon as it starts, I am prompted for root password, but l ways get back an __Elevation error__ :
`User did not grant permission`
(which I did)

[image]

jviotti, 2016-04-12:
@loranger That's strange, I'm thinking you could have some weird `sudo` configuration, or you're hitting an edge case on [sudo-prompt](https://github.com/jorangreef/sudo-prompt), the module that we use to handle elevation.

Can you try running the `.app` with `sudo` directly?

```sh
sudo open path/to/Etcher.app
```

Does that work?

loranger, 2016-04-12:
Thanks for you reply, @jviotti 

I'm afraid it doesn't work either.
As [you can see](https://dl.dropboxusercontent.com/s/57heeky2zj0s22t/screencast_2016-04-12_15-50-01.mp4?dl=0), as I type your command and give my password, Etcher is launching (at least in the Dock), then prompt me again for my password. I give it again, and then get the error message above.

jviotti, 2016-04-12:
Hi @loranger 

This is indeed something I never saw before. The fact that even explicitly running the `.app` with `sudo` insists in elevation is a red sign.

Can you the following gist: https://gist.github.com/jviotti/a97604a30c5e025430930e14c15f4e4d?

Assuming you have [NodeJS](https://nodejs.org/en/) installed:

```sh
cd path/to/gist
npm install
```

And give it a go with and without `sudo`:

```sh
node app.js
sudo node app.js
```

I get `false` and `true`, respectively, in OS X 10.11.4.

loranger, 2016-04-12:
I get the same result as your :

```
Merlin ~/D/d/a97604a30c5e025430930e14c15f4e4d ⎠⎠⎠ la                                                    ⏎ master
total 16
drwxr-xr-x  12 loranger  staff   408B 12 avr 17:31 .git
-rw-r--r--   1 loranger  staff    56B 12 avr 17:30 app.js
-rw-r--r--   1 loranger  staff   295B 12 avr 17:30 package.json
Merlin ~/D/d/a97604a30c5e025430930e14c15f4e4d ⎠⎠⎠ npm install                                             master
is-root-test@1.0.0 /Users/loranger/Desktop/debug_etcher/a97604a30c5e025430930e14c15f4e4d
└── is-root@1.0.0

npm WARN is-root-test@1.0.0 No description
npm WARN is-root-test@1.0.0 No repository field.
Merlin ~/D/d/a97604a30c5e025430930e14c15f4e4d ⎠⎠⎠ node app.js                                           master ◼
false
Merlin ~/D/d/a97604a30c5e025430930e14c15f4e4d ⎠⎠⎠ sudo node app.js                                      master ◼
Password:
true
Merlin ~/D/d/a97604a30c5e025430930e14c15f4e4d ⎠⎠⎠                                                       master ◼
```

(It sounds good, isn't it ?)

jviotti, 2016-04-13:
@loranger Yeah, looks good. I suspect this might be an issue with the module we're using to provide the elevation dialog (`sudo-prompt`).

Can you try running the following gist now and see what it outputs?

jviotti, 2016-04-13:
Another good experiment to make would me replacing the `whoami` command in that gist for `node -e "console.log(process.getuid())"`, and see what it outputs as well.

Notice that if you ran the dialog once and it succeeded, the sudo session will be cached and you might get inaccurate results. I recommend running `sudo -k` each time before running the gists.

loranger, 2016-04-13:
I'm sorry but the last gist does not work :
```
Merlin ~/D/d/d47c1f20dddd7337d1a672af838e2a88 ⎠⎠⎠ npm install                                             master
sudo-prompt-test@1.0.0 /Users/loranger/Desktop/debug_etcher/d47c1f20dddd7337d1a672af838e2a88
└── sudo-prompt@2.3.0

npm WARN sudo-prompt-test@1.0.0 No description
npm WARN sudo-prompt-test@1.0.0 No repository field.
Merlin ~/D/d/d47c1f20dddd7337d1a672af838e2a88 ⎠⎠⎠ node app.js                                           master ◼
/Users/loranger/Desktop/debug_etcher/d47c1f20dddd7337d1a672af838e2a88/app.js:7
    throw error;
    ^

Error: User did not grant permission.
    at /Users/loranger/Desktop/debug_etcher/d47c1f20dddd7337d1a672af838e2a88/node_modules/sudo-prompt/index.js:21:38
    at ChildProcess.exithandler (child_process.js:209:5)
    at emitTwo (events.js:100:13)
    at ChildProcess.emit (events.js:185:7)
    at maybeClose (internal/child_process.js:850:16)
    at Socket.<anonymous> (internal/child_process.js:323:11)
    at emitOne (events.js:90:13)
    at Socket.emit (events.js:182:7)
    at Pipe._onclose (net.js:475:12)
Merlin ~/D/d/d47c1f20dddd7337d1a672af838e2a88 ⎠⎠⎠                                                     ⏎ master ◼
```
The graphical sudo did prompt, and I filled it, but it ends with this error.

Then I tried with the `node -e` and got the same error :
```
Merlin ~/D/d/d47c1f20dddd7337d1a672af838e2a88 ⎠⎠⎠ cat app.js                                        ⏎ master ✱ ◼
var sudo = require('sudo-prompt');

sudo.exec('node -e "console.log(process.getuid())"', {
  name: 'Sudo Prompt Test'
}, function(error, stdout, stderr) {
  if (error) {
    throw error;
  }

  console.log('STDOUT: ' + stdout);
  console.error('STDERR: ' + stderr);
});
Merlin ~/D/d/d47c1f20dddd7337d1a672af838e2a88 ⎠⎠⎠ node app.js                                         master ✱ ◼
/Users/loranger/Desktop/debug_etcher/d47c1f20dddd7337d1a672af838e2a88/app.js:7
    throw error;
    ^

Error: User did not grant permission.
    at /Users/loranger/Desktop/debug_etcher/d47c1f20dddd7337d1a672af838e2a88/node_modules/sudo-prompt/index.js:21:38
    at ChildProcess.exithandler (child_process.js:209:5)
    at emitTwo (events.js:100:13)
    at ChildProcess.emit (events.js:185:7)
    at maybeClose (internal/child_process.js:850:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
Merlin ~/D/d/d47c1f20dddd7337d1a672af838e2a88 ⎠⎠⎠                                                     ⏎ master ◼
```

loranger, 2016-04-13:
By the way, here is my environment :

**node version**: v5.10.1
**npm version**: 3.8.6
**platform**:  Darwin Merlin 15.4.0 Darwin Kernel Version 15.4.0: Fri Feb 26 22:08:05 PST 2016; root:xnu-3248.40.184~3/RELEASE_X86_64 x86_64 (OS X 10.11.4)

jviotti, 2016-04-13:
Hi @loranger ,

That's great news. At least we know the issue is in [sudo-prompt](https://github.com/jorangreef/sudo-prompt). I'm not very familiar on the internal workings on this module for OS X, but @jorangreef, the author, might be able to provide some insight.

I encourage you to create a new issue in that module, explaining what happens, describing your platform and linking back to this issue, so I can stay involved and hopefully be able to help further.

jorangreef, 2016-04-13:
Thanks @loranger, would you please test the following on that machine:

1. Open a Terminal window
2. `sudo -k`
3. `sudo mkdir -p /var/db/sudo/$USER`
4. `sudo touch /var/db/sudo/$USER`
5. Quit the Terminal window
6. Open another Terminal window
7. `sudo touch /var/db/sudo/$USER`

Does step 7 ask for your password?

loranger, 2016-04-14:
@jviotti Thanks for your support, I've created [a ticket](https://github.com/jorangreef/sudo-prompt/issues/17) in the [sudo-prompt](https://github.com/jorangreef/sudo-prompt) repository.

@jorangreef Thank you for digging in my issue. I did run your test, and yes the step 7 did ask for my password. Is my issue related to the [OS X 10.10.5 kernel local privilege escalation](https://news.ycombinator.com/item?id=10067524) issue I did fix by adding a `Defaults tty_tickets` in my `/etc/sudoers` ?

jorangreef, 2016-04-14:
Yes, if step 7 asked for your password it's because `tty_tickets` are enabled. This is an issue in `sudo-prompt` (https://github.com/jorangreef/sudo-prompt/issues/13). I hope to have support for `tty_tickets` on OS X in the next few days.

Thanks @loranger and @jviotti for your help debugging this.

jviotti, 2016-04-14:
Awesome, thanks for the great help @jorangreef. Let me know if there is anything I can collaborate with on that issue.

jorangreef, 2016-04-15:
Thanks @jviotti, I made a few major changes in https://github.com/jorangreef/sudo-prompt/commit/0b5c7d736188643a37250f56bdeaf82a356e69e8, would you mind testing on OS X (with and without tty_tickets) and Linux? It's not yet published on NPM.

jviotti, 2016-04-15:
Hi @jorangreef,

Thanks for the quick fix. I've tested this in OS X. It works fine with and without `tty_tickets`, however I can't pass relative paths to `sudo.exec` anymore given they will resolve to a new location under `/private/`.

[image]

Maybe you can preserve the current working directory?

I don't have access to a Linux box atm, but will post back my results probably later today.

jviotti, 2016-04-15:
I borrowed an Ubuntu 14.04 AMD64 laptop, and I confirm it works fine there as well.

jorangreef, 2016-04-16:
Thanks @jviotti, I added a commit to preserve the cwd on OS X. It may or may not always be the case on Linux. I would like to test the various binaries on Linux further in future to address this properly. Perhaps we can use the same applet.app sub-shell approach for Linux that we are now using on OS X (see sudo-prompt-script within the MacOS directory of the applet). That should make it possible to support setting custom ENV variables and cwd via options.

jviotti, 2016-04-18:
@jorangreef I confirm OS X is working fine with relative paths again. The sub-shell approach makes sense for Linux as well. We don't use `sudo-prompt` for Linux over here so I guess we're safe to update once you publish, but feel free to ping me if you want me to test any Linux functionality (I have two Ubuntu 14.04 boxes and a Fedora 22).

closed 2016-04-26 by jviotti