balena-io/etcher #298: Fix NPM graceful-fs deprecation warnings

jviotti, 2016-04-12:
```sh
npm WARN deprecated graceful-fs@3.0.8: graceful-fs version 3 and before will fail on newer node releases. Please update to graceful-fs@^4.0.0 as soon as possible.
```

jviotti, 2016-04-12:
`ack` results:

```
$ ack --noignore-dir=node_modules --json graceful\\\-fs\\\"\\\:
node_modules/electron-builder/node_modules/electron-packager-tf/node_modules/asar/node_modules/mksnapshot/node_modules/decompress-zip/package.json
41:    "graceful-fs": "^3.0.0",

node_modules/electron-builder/node_modules/electron-packager-tf/node_modules/asar/node_modules/mksnapshot/node_modules/fs-extra/package.json
44:    "graceful-fs": "^3.0.5",

node_modules/electron-builder/node_modules/fs-extra/package.json
39:    "graceful-fs": "^4.1.2",

node_modules/electron-builder/node_modules/meow/node_modules/normalize-package-data/test/fixtures/npm.json
41:    "graceful-fs": "~1.2.0",

node_modules/electron-builder/node_modules/meow/node_modules/normalize-package-data/test/fixtures/read-package-json.json
26:    "graceful-fs": "~1.1.8"

node_modules/electron-builder/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/load-json-file/package.json
34:    "graceful-fs": "^4.1.2",

node_modules/electron-builder/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/path-type/package.json
42:    "graceful-fs": "^4.1.2",

node_modules/electron-builder/node_modules/read-package-json/node_modules/normalize-package-data/test/fixtures/npm.json
41:    "graceful-fs": "~1.2.0",

node_modules/electron-builder/node_modules/read-package-json/node_modules/normalize-package-data/test/fixtures/read-package-json.json
26:    "graceful-fs": "~1.1.8"

node_modules/electron-builder/node_modules/read-package-json/package.json
22:    "graceful-fs": "^4.1.2"
29:    "graceful-fs": "^4.1.2"

node_modules/electron-mocha/node_modules/fs-extra/package.json
39:    "graceful-fs": "^4.1.2",

node_modules/electron-mocha/node_modules/mocha/node_modules/glob/package.json
20:    "graceful-fs": "~2.0.0",

node_modules/electron-packager/node_modules/asar/node_modules/mksnapshot/node_modules/decompress-zip/package.json
36:    "graceful-fs": "^4.1.3",

node_modules/electron-packager/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/normalize-package-data/test/fixtures/npm.json
41:    "graceful-fs": "~1.2.0",

node_modules/electron-packager/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/normalize-package-data/test/fixtures/read-package-json.json
26:    "graceful-fs": "~1.1.8"

node_modules/electron-packager/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/load-json-file/package.json
34:    "graceful-fs": "^4.1.2",

node_modules/electron-packager/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/path-type/package.json
42:    "graceful-fs": "^4.1.2",

node_modules/electron-packager/node_modules/fs-extra/package.json
39:    "graceful-fs": "^4.1.2",

node_modules/electron-prebuilt/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/normalize-package-data/test/fixtures/npm.json
41:    "graceful-fs": "~1.2.0",

node_modules/electron-prebuilt/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/normalize-package-data/test/fixtures/read-package-json.json
26:    "graceful-fs": "~1.1.8"

node_modules/electron-prebuilt/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/load-json-file/package.json
34:    "graceful-fs": "^4.1.2",

node_modules/electron-prebuilt/node_modules/electron-download/node_modules/nugget/node_modules/pretty-bytes/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/path-type/package.json
42:    "graceful-fs": "^4.1.2",
[... 42 lines cut ...]

node_modules/gulp-jshint/node_modules/gulp-util/node_modules/dateformat/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/path-type/package.json
42:    "graceful-fs": "^4.1.2",

node_modules/gulp-sass/node_modules/gulp-util/node_modules/dateformat/node_modules/meow/node_modules/normalize-package-data/test/fixtures/npm.json
41:    "graceful-fs": "~1.2.0",

node_modules/gulp-sass/node_modules/gulp-util/node_modules/dateformat/node_modules/meow/node_modules/normalize-package-data/test/fixtures/read-package-json.json
26:    "graceful-fs": "~1.1.8"

node_modules/gulp-sass/node_modules/gulp-util/node_modules/dateformat/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/load-json-file/package.json
34:    "graceful-fs": "^4.1.2",

node_modules/gulp-sass/node_modules/gulp-util/node_modules/dateformat/node_modules/meow/node_modules/read-pkg-up/node_modules/read-pkg/node_modules/path-type/package.json
42:    "graceful-fs": "^4.1.2",

node_modules/gulp-sass/node_modules/node-sass/node_modules/gaze/node_modules/globule/node_modules/glob/package.json
16:    "graceful-fs": "~1.2.0",

node_modules/gulp-sass/node_modules/node-sass/node_modules/node-gyp/package.json
27:    "graceful-fs": "^4.1.2",

node_modules/resin-zip-image/node_modules/unzip2/node_modules/fstream/package.json
19:    "graceful-fs": "~3.0.2",
```

jviotti, 2016-04-13:
The culprit of these errors are dependencies, which are already up to date. Its their responsibility to update, which they will eventually, so safe to close this.

closed 2016-04-13 by jviotti

imrehg, 2016-08-03:
@jviotti might not be this simple as the closing comment

Currently this is the affected dependency tree:
```
├─┬ president@2.0.3
│ └─┬ windosu@0.2.0
│   └─┬ temp@0.6.0
│     └─┬ rimraf@2.1.4
│       └── graceful-fs@1.2.3 
├─┬ unzip2@0.2.5
│ └─┬ fstream@0.1.31
│   └── graceful-fs@3.0.8 
└─┬ update-notifier@0.6.3
  └─┬ configstore@2.0.0
    └── graceful-fs@4.1.5 
```

Affected modules:
* `windosu`: filed issue, as the dependencies below were updated https://github.com/tehsenaus/windosu/issues/13 (so it's good, probably just add time)
* `unzip2`: last updated 2 years ago, and people discussing this a while in the issue tracker, might need to find a replacement or fork? https://github.com/glebdmitriew/node-unzip-2/issues/13 (blocking)

jviotti, 2016-08-03:
Hi @imrehg ,

Thanks for the research on this ticket. We recently replaces `unzip2` with `yauzl` (https://github.com/thejoshwolfe/yauzl), so I think we're good on that.

I'll definitely upgrade `windosu` once your issue gets addressed.

imrehg, 2016-08-03:
@jviotti actually suddenly I realized, I am being stupid - that dependency tree was from the resin-cli, so replied to the wrong repo :/ Thanks for your patience, and glad that it sort of applies as well.
Though it also means resin-cli needs an update to `yauzl` too...

jviotti, 2016-08-03:
Hi @imrehg ,

> Though it also means resin-cli needs an update to `yauzl` too...

Yeah, `unzip2` will go away once I upgrade `etcher-image-write` to the latest version in the `resin-cli`.