balena-io/etcher #304: Prevent drive from auto-selecting when the drive modal selector is opened

jviotti, 2016-04-12:
Consider the following situation:

- You have two connected drives.
- You open the drive selector modal by clicking "Select Drive".
- Both drives appear on the modal.
- You eject one modal.

In this case, the modal remains opened with a single drive, but in the background, the drive has already been auto-selected.

A good solution is to prevent auto-selection while the modal is opened.

alexandrosm, 2016-04-12:
But why is that a problem? The drive selection will presumably change if
the modal is submitted. I think the modal should just reflect the
auto-selection.

jviotti, 2016-04-12:
@alexandrosm That's an even better solution. Currently, the modal wouldn't update if there were changes in the background. I was attempting to solve a symptom, but your suggestion led me to fix the root cause.

closed 2016-04-12 by jviotti