balena-io/etcher #307: Uncaught error and program unable to terminate

mdnt2mdnt, 2016-04-12:
[image]
[image]

vankasteelj, 2016-04-13:
Exact same error on Ubuntu 15.10 amd64

jviotti, 2016-04-13:
Hi @mdnt2mdnt , @vankasteelj 

> Note that this error while creating a bootable usb with Ubuntu 15.10 amd64 iso doesn't prevent the usb from being created, and it's bootable on my Z87 UEFI motherboard

The error seems to happen during the validation step, so the drive was burnt correctly. What surprises me is that `stream.push() after EOF` implies we're trying to push data to a closed stream, which is not the case (the validation process only reads data back from the device).

***

Can you actively reproduce this with any image? If not, do you mind sharing the image that causes the error? Also, does the error go away if you try writing the same image to a different USB Drive/SDCard?

vankasteelj, 2016-04-13:
It's ubuntu-15.10-desktop-amd64.iso from http://releases.ubuntu.com/15.10/

jviotti, 2016-04-13:
Good news, I can reproduce with that image. I'll do some investigation a push a fix soon. Thanks a lot for the help so far!

jviotti, 2016-04-14:
Hi @mdnt2mdnt and @vankasteelj 

I've been the whole day tracing down this bug, and it took me to [slice-stream](https://github.com/EvanOxfeld/slice-stream), which is a module we use to perform the validation. This module, as the name implies, allows us the read a stream up to a certain point and ignore the rest (basically an `Array.slice(0, size)`). We use this to perform validation since, for example, if you write a 2GB image to an 6GB, in order to validate the write, we have to read back from the drive exactly 2GB, otherwise we're reading unwritten data that skews the checksum.

There seems to be a very subtle issue in this `slice-stream` module that causes one more buffer to be read even when you already signalled the stream to close, which happens in some cases, like in this Ubuntu image. 

The good news is that I have a patch for `slice-stream` that looks like it solves the issue, but I'm still trying to figure out why this is the case, and why this happens with only certain images.

I'll post back my results, and probably ask you to test the fix. Thanks a lot for your patience and support!

jviotti, 2016-04-14:
Hi @mdnt2mdnt and @vankasteelj ,

I've submitted a PR to `slice-stream`: https://github.com/EvanOxfeld/slice-stream/pull/2.

You can check the PR for more information, but basically what happens is that the image size is not cleanly divisible by the block size your operating system is using to read it, causing one last block to be processed even after the end of the stream has been signalled.

jviotti, 2016-04-14:
Uhm, that package was last updated in 2 years ago, so I doubt its still maintained. Let's wait for a couple of days, and I'll maintain a fork otherwise.

closed 2016-04-15 by jviotti