balena-io/etcher #308: internationalization (i18n)

vankasteelj, 2016-04-13:
I suggest you to use an internationalization engine like i18n (angular supports it), in order for people to translate the strings and texts in the app.

I can't help you with setting i18n up, but you can ping me for French localisation.

jviotti, 2016-04-13:
Hi @vankasteelj 

This is a great suggestion. I'm familiar with Angular's internationalisation support, so I should be able to configure it. We'll ping you for help with the French localisation for sure!

levrik, 2016-08-24:
I could hell with a German translation.

RazrSaw, 2016-11-06:
I can help with Hungarian translation.

crushedice2000, 2017-01-29:
I can help with the Spanish translation

PrzemekSkw, 2017-02-27:
I could help with Polish tranlation

gustavohmsilva, 2017-04-11:
I can maintain the Brazilian Portuguese translation.

TotalCaesar659, 2017-05-14:
I can help with Russian translation.
P.S. How can I translate shortcut comment (/usr/share/applications/etcher-electron.desktop) in this repo? Here is no file .desktop or something like this.

> [Desktop Entry]
> Name=Etcher
> Comment=Flash OS images to SD cards & USB drives, safely and easily.
> **Comment[ru]=Легкая и безопасная запись образов систем на SD-карты и USB-флэшки.**
> GenericName=Etcher
> Exec=etcher-electron %U
> Icon=etcher-electron
> Type=Application
> StartupNotify=true
> Categories=Utility;

lurch, 2017-05-15:
Etcher doesn't have any support for localisation (yet).
If you're using the AppImage, the `.desktop` file is created by https://github.com/resin-io/etcher/blob/master/scripts/build/electron-create-appdir.sh but if you're using the debian package, the `.desktop` file is created by `electron-installer-debian` (see https://github.com/resin-io/etcher/blob/master/scripts/build/electron-installer-debian-linux.sh and https://github.com/resin-io/etcher/blob/master/scripts/build/debian/config.json )

Jacalz, 2018-03-07:
I could definitely go ahead and help with Swedish translations when this gets added :+1:

yarons, 2018-03-12:
Hebrew is on board, we may also need some assistance with adding RTL support to the UI (For the sake of Hebrew, Arabic, Persian, etc.).

Vistaus, 2018-03-18:
I want to translate Etcher to Dutch. I hope localization support gets added soon, esp. now that it's getting really popular it's important that the app can be translated.

tuhaihe, 2018-05-17:
I could help with a Chinese(Simplified) translation.

KovalevArtem, 2019-07-18:
Is there any news about the localization of the program?

thundron, 2019-07-22:
Not yet, we're doing some heavy lifting wrt upgrading the UI code which will surely help with localization once we're done

EldersJavas, 2020-04-30:
Sorry, my English is poor.
Hello, everyone, I'd suggest that this project use the transifex for localization. All it takes is an XML file or some other format for the interpreter to translate into any language, so if we use transifex, our workload will be significantly reduced.
This is a little bit of advice from me,

KovalevArtem, 2020-04-30:
My opinion is that Transifex is the most inconvenient translation service.
I would recommend using Weblate or Crowdin...

@orangesunny of Weblate can help set everything up.
Gratis for libre projects...

Vistaus, 2020-04-30:
> My opinion is that Transifex is the most inconvenient translation service.
> I would recommend using Weblate or Crowdin...
> 
> @orangesunny of Weblate can help set everything up.
> Gratis for libre projects...

Funny. In my opinion, Crowdin is the worst translation service together with OneSky. I like Transifex the best, but Weblate is also pretty good.

Porrumentzio, 2020-05-25:
Is there any news about this. Both Transifex and Weblate would be good, or if not even .pot/.po/.mo files would be good.
I would make a basque translation, and maybe esperanto too.

orangesunny, 2020-06-14:
Well, I want to break the silence. We in Weblate will be happy to host the translations. Just let us know on https://hosted.weblate.org/hosting/ if you want to translate :) 

And thanks for the Etcher! It made a reinstallation of a laptop nicer experience for me a few days ago :)

Porrumentzio, 2020-08-05:
Is there any news about this?

shawaj, 2020-12-13:
Also requested in #3044

RoderickQiu, 2022-07-16:
Etcher is good and **I think I might try to contribute a i18n part in July or August as well as contributing a Simp Chinese support for it**.