balena-io/etcher #312: Creating bootable USB corrupts the key

vankasteelj, 2016-04-13:
I have a 16GB USB dongle, classic. Using other tools to load a ubuntu image onto it creates a 16GB partition with about 1.5GB used, and it's called "Ubuntu <version><arch>" or similar. 

Here, it "seems" to do the same, but it doesn't:

- `lsblk -l`:
```
  sdd    8:48   1  14,6G  0 disk 
  sdd1   8:49   1   1,1G  0 part /media/vankasteelj/Ubuntu 15.10 amd64
  sdd2   8:50   1   2,2M  0 part 
```

- gparted:
[image]

(keep in mind it's a 16GB key, not 64 like etcher seemed to write it. 

- libparted log:
```
Driver descriptor indicates a physical bloc size of 2048 bytes, but Linux says 512 bytes.
```

jviotti, 2016-04-13:
Hey there, 

The module that performs the actual writing is [resin-image-write](https://github.com/resin-io-modules/resin-image-write). Notice it has an `example.js` in the root of the project.

Can you try running it against this image and drive and see what you get?

> Driver descriptor indicates a physical bloc size of 2048 bytes, but Linux says 512 bytes.

Not sure if it will make any difference, but you could try tweaking the `CHUNK_SIZE` in `resin-image-write` and set it to what Linux expects: https://github.com/resin-io-modules/resin-image-write/blob/master/lib/write.coffee#L32

***

I'm downloading this Ubuntu image and will test myself in a minute.

vankasteelj, 2016-04-13:
To recover the usb key (in order to have it working like it should, recognized across OSes), I had to:

1) Overwrite the partition table:
```
sudo dd if=/dev/zero of=/dev/sdX bs=512 count=1
```

2) Recreate a msdos partition and a FAT32 partition with a label (using gparted)

jviotti, 2016-04-15:
@vankasteelj Do you mind trying to burn this image again with the latest changes from `master`? At least here in OS X, looks like the issue was gone along with this fix: https://github.com/resin-io/etcher/issues/307.

vankasteelj, 2016-04-15:
I can try, but It'll sure mess my usb key MBR again, right?

jviotti, 2016-04-15:
@vankasteelj When I was debugging the EOF error by repeatedly burning the Ubuntu Desktop image the drive was never readable. After the fix, the drive was mounting correctly saying "Ubuntu 15.10" or something like that. I didn't inspect it in depth, but the fact that it mounted with the correct label means that the MBR got read. 

This was on OS X though, I'll try on Linux on Monday, but if you pull the latest changes from the repo, and 'npm install' to get the latest version of the dependency, I believe it should work.

vankasteelj, 2016-04-16:
yeah on linux it was read without any issue too, but it's on windows (and with linux deeper inspection) that I noticed the completely messed up MBR and partition table.

(nb: and the fact that unix* platforms can read those kind of messed up partitions is what makes them great, but the evidence remains that most ppl use windows)

jviotti, 2016-06-30:
Hi @vankasteelj ,

I've re-tested this recently and couldn't reproduce anymore. Can you confirm this is the case for you as well with the latest Etcher version (v1.0.0-beta.10)?

jviotti, 2016-07-20:
We've been investigating this issue in much more depth and hopefully got to the bottom of it.

Ubuntu uses a hybrid format on their ISOs so they can be directly burnt to CDs and USB drives and boot without any problems, however this "hack" confuses `parted` a lot, which interprets the GPT to be invalid, and instead thinks the partition table is "mac", eventually throwing the warning you see there.

The behaviour you're experiencing can be reproduced with `dd`, and its not an issue on Etcher. The results are indeed very strange, but they are safe to ignore.

Some resources:

- https://lists.ubuntu.com/archives/ubuntu-devel/2011-June/033495.html
- https://blog.einval.com/2011/01/07#isohybrid_CDs

We'll add some documentation about this though, so we have a place to point users that run `parted` on it, and think there is a problem in the Etcher/image side.

closed 2016-07-20 by jviotti

vankasteelj, 2016-07-20:
"safe to ignore", you'll have to say that quick, it does mess up the partition table and render the usb key "broken" for people not knowing how to fix it :/

alexandrosm, 2016-07-20:
But the behaviour is same for dd. Do you have any particular reason to
believe this should be an etcher reason, or any suggestions as to how
etcher should fix this issue? Ultimately, Etcher writes what you tell it to
the SD card. If the image is problematic, the sd card will be problematic.
We're open to ideas, but I don't see how Etcher can do something
differently in this case.

vankasteelj, 2016-07-20:
LiLi, Unetbootin and others manage to do it

jviotti, 2016-07-21:
The programs that you mention are specifically made to deal with GNU/Linux images and operate in a completely different way. They format the drive, copy the contents, mark the partition as bootable and install GRUB; while Etcher simply copies byte by byte the image to the drive (it flashes the image).

Drives flashes with Ubuntu exhibit this behaviour since Ubuntu images contains basically a "hack" so that they can be simply flashed and booted from without the extra manipulation that other programs perform.

jviotti, 2016-07-21:
Etcher doesn't mess the partition table in any way. That's just a minor consequence of Ubuntu hacks, which enable the image to be bootable in a whole range of systems.

wayofthefuture, 2017-04-22:
I can confirm the problem still occurs. Ubuntu 17.04 exactly like the image above in the first post.

lurch, 2017-04-22:
Yes, Ubuntu 17.04 is still doing the same hybrid ISO hacks https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

wayofthefuture, 2017-04-22:
Not to mention it locks 1 of the partitions too. After you etch a usb good luck getting it back to unallocated space.

alexandrosm, 2017-04-22:
Once again, etcher will directly copy the bits you give it onto the block
device. If you want a program that does something different, use a
different program.

jviotti, 2017-04-22:
> After you etch a usb good luck getting it back to unallocated space.

That's a trivial task in all OSes. We have some command line instructions in the docs: https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

magicvince, 2018-10-29:
Same problem using etcher on Debian Stretch with Debian's ISO (original or remastered  Debian Stretch, LMDE3, Primtux ...) on different usb stick (brand and space). It works to copy and boot the image but if you want to use gparted for partitioning the HD, "Driver descriptor indicates a physical bloc size of 2048 bytes, but Linux says 512 bytes.".
It is impossible to make any operation on the stick (copy or delete a file for exemple)
It was also impossible to reuse the stick if I don't use the sudo dd if=/dev/zero of=/dev/sdX bs=512 count=1 et build a new ms-dos table of partition.