balena-io/etcher #325: Etcher to open .GZ files

kkaiser1952, 2016-04-14:
When I burn the SD to a hard drive I generally save it as .GZ for space reasons. It would be great if Etcher recognized these and could use them to burn it back to an SD.

jviotti, 2016-04-14:
Hi @kkaiser1952 ,

So you mean Etcher should be able to burn `image.img.gz` or `image.iso.gz` right? If so, I think this is a good addition, but I wonder how this will plan with the fact that Etcher currently writes ZIPs that containing a single image file inside, but are not necessarily a "zipped" single file.

I think checking the extension should be a good enough clue.

kkaiser1952, 2016-04-14:
Well if what you are saying is to let Etcher open a .GZ file from my desktop and burn a SD card with it then yes, thats what I mean.

noglitch, 2016-04-19:
@jviotti I've just tried Etcher and it seems fantastic for what we put in place with our board SD images.
But while you are at it, I would also suggest to add .bz2 images as well.
In fact we produce .img.bz2 images for SD card right now: http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d2XplainedMainPage#Demo_archives

To write them on a Windows platform, we advice users to use the Rawrite32 tool as it handles such a file format...

Thanks, bye.

jviotti, 2016-04-19:
@noglitch Thanks for the suggestion! I edited the title to include `.bz2` support.

RobertCNelson, 2016-04-20:
@jviotti fantastic tool, if you include *.img.xz files beagleboard.org can switch too... ;)

jviotti, 2016-04-20:
@RobertCNelson Thanks a lot! Taking notes :)

***

To summarise:

- [ ] Support for `.gz`
- [ ] Support for `.bz2`
- [ ] Support for `.7z`
- [ ] Support for `.xz`

jviotti, 2016-05-04:
Hi @RobertCNelson ,

I've just published v1.0.0-beta.5 with XZ support. Let us know what you think!

jmark1m, 2016-05-18:
XZ support is great but please don't stop there.  The GZ support is helpful because GZ decompresses so much faster the XZ.  
http://catchchallenger.first-world.info/wiki/Quick_Benchmark:_Gzip_vs_Bzip2_vs_LZMA_vs_XZ_vs_LZ4_vs_LZO

jviotti, 2016-05-18:
Hi @jmark1m ,

Definitely, we'll provide support for all compression formats discussed above. We're currently prioritising auto-updating functionality, and we'll probably get all the formats out in a row afterwards.

jviotti, 2016-05-19:
Sorry for the huge delay on this issue, I'm finally allocating some time to finish the remaining compression formats. I just PRed GZ support, and will be looking forward to PR bzip2 support as well today.

closed 2016-05-24 by jviotti

noglitch, 2016-05-27:
Hi

I've just tested the xz and bz2 image writers with 1.0.0-beta7 version and:
- xz image flasher works great
- I can't use the .bz2 format:
  * I can't select a file with this format unless there is a file with the same name but a .zip extension in the directory
  * when I select the img.bz2 file, the .zip file is presented instead in the GUI, just underneath the "select image icon"
  * when I try to start flashing... it doesn't go further with sometimes a window saying "an error occurred". Note, when etcher fails like this, I have big difficulties to actually kill it...

(I run it from Linux Ubuntu 16.04)

Hope that it can help, Bye,

jviotti, 2016-05-30:
Hi @noglitch ,

Thanks for reporting this! It seems that you're hitting this error: https://github.com/regular/unbzip2-stream/issues/7 from one of the dependencies we're using. I'll try to allocate some time to work myself on it!

jviotti, 2016-07-19:
Hi @noglitch ,

Turns out the bz2 selection issue was an Electron bug with the open dialog on GNU/Linux (see https://github.com/electron/electron/issues/6305), which was fixed in the latests versions!

jviotti, 2016-09-12:
Hi @noglitch ,

I'm happy to announce the bz2 issue you've experienced has just been fixed in the bz2 implementation we're using (se https://github.com/resin-io-modules/etcher-image-stream/pull/37), and it's coming on the next release!

mpvader, 2017-04-26:
Hi, running Etcher 1.0.0.0-rc4, I get this error:

[image]

But opening the same file in 7zip works fine.

Running Windows 8.1 Pro.

jhermsmeier, 2017-04-26:
Hey @mpvader, that's because Etcher currently checks the image extension – in this case `.rpi-sdimg`, and bails out on that because it's not mapped as a known format.
We're currently working on making it fall back to treating anything unknown as if it had an `.img` extension – so, renaming your file to `venus-image-rapsberrypi2.img.bz2` should do the trick for now.

mpvader, 2017-04-26:
Hi @jhermsmeier, thank you for the fast response. Should it just compare the most right part? So as long as it ends in .bz2 it's fine?

These 'cascading' extensions are quite common for images and such.

mpvader, 2017-04-26:
I meant shouldn't

mpvader, 2017-04-26:
And, just to be clear: I'd expect etcher to unzip the .bz2 file, and burn its contents. Not burn the zipped file.

jhermsmeier, 2017-04-26:
@mpvader well, what it basically does is look at the base extension (i.e. `.bz2` here), and then checks if it knows how to deal with the thing that has been compressed (in this case the next extension, `.rpi-sdimg`) – so what comes before the `.bz2` does factor into how Etcher treats it.

> And, just to be clear: I'd expect etcher to unzip the .bz2 file, and burn its contents. Not burn the zipped file.

Yup, that's exactly what it does (unzip & flash the contained image)

mpvader, 2017-04-26:
ahh.. thanks. Understood! looking forward to your fix then :)

lurch, 2017-05-03:
@jhermsmeier Was this fixed by #1363 ?

jhermsmeier, 2017-05-03:
Only partially – there's another check in the GUI, which won't let you select unrecognised images – working on fixing that now.

lurch, 2017-05-03:
But presumably that GUI-check will still allow @mpvader 's `venus-image-rapsberrypi2.rpi-sdimg.bz2` file to be selected?

jhermsmeier, 2017-05-03:
@lurch oh yes, right, that should work already now