balena-io/etcher #326: Middle button should change when SD card removed

geomic, 2016-04-15:
When removing an SD card, you need to click middle button (after removing USB drive) to get Etcher to realise the drive is missing now.

That's on Mac OS X Yosemite.

jviotti, 2016-04-15:
Hey there @geomic ,

Nice to see you playing around with Etcher :) What do you mean by click middle button? So when the drive selector modal is open, you have to click somewhere for it to update?

Maybe you can record a little video?

geomic, 2016-04-22:
@jviotti this appears fixed, as of the latest version. Thanks!

Essentially, when plugging an SD card in, the drive is identified correctly.

- When unplugging the SD card on the version of Etcher I had about a week ago, the reference to /dev/diskX in the middle wouldn't change. **Clicking** there, would correctly change to the "Connect a drive button", but it wouldn't happen automatically for some reason.

- Now, when unplugging the SD card on the latest version, the interface correctly automatically changes to the red "Connect a drive" button, albeit showing a "null null" label first for about a second.

closed 2016-04-22 by geomic