balena-io/etcher #332: Add more command line output (verbose/debug output and -h/--help)

mccollam, 2016-04-15:
In trying to debug a hang (which appears to be already filed as issue #307) I was unable to see any information about what Etcher was doing internally at any step.  I expected a "-v" or "--verbose" switch (or similar) to enable this but that did not work.

Additionally, running Etcher with "-h" or "--help" to find the list of supported command line switches produced no output, which is unusual for a tool of this kind.

jviotti, 2016-04-15:
Hi @mccollam, what operating system are you running?

The problem with the output is that to perform elevation on startup, the app first runs without privileges, and then spawns a process to run itself again with privileges once you provide your password to the sudo dialog.

Sadly the output streams are not inherited by the children process (maybe there is a way to do it on `sudo-prompt`, but I wonder how that would play in Windows, or OS X because of `open`), so you get no output.

I'm planning to add a command line option to omit the elevation dialog, so you can take care of it yourself, and thus prevent the app from forking a new process.

There is a fix coming for the issue you linked though, if you're interested, I can ping you when it lands to `master` so you can check yourself.

mccollam, 2016-04-15:
This is on Ubuntu 15.10 x86_64.

I don't see the behavior you are describing -- if I launch as a normal user I just get a dialog box telling me to rerun as root.

[image]

I was planning to try to submit a simple patch to provide -h at least but if it's already in the works I'll hold off. :)

jviotti, 2016-04-15:
Hi @mccollam ,

Yeah, GNU/Linux doesn't yet have the elevation dialog (see https://github.com/resin-io/etcher/issues/277) sadly, but in GNU/Linux you should be able to see logging. Since Electron runs in two processes: the main and renderer processes, things logged in one will not appear in the other. I believe this is the issue you have, since the `stream.push() after EOF` only happens in the renderer process. Notice you can open DevTools by pressing Ctrl+Alt+I.

> I was planning to try to submit a simple patch to provide -h at least but if it's already in the works I'll hold off. :)

Sounds great! By "working on it", I mean its planned, but work didn't start yet. I'd be awesome to see a patch :)

jviotti, 2016-04-15:
BTW, I just merged the `stream.push() after EOF` fix to `master`, which is actually implemented in [slice-stream](https://github.com/jviotti/slice-stream2), which is a dependency of [resin-image-write](https://github.com/resin-io-modules/resin-image-write), which is a dependency of Etcher :)

The fix commit, in case you want to know more: https://github.com/jviotti/slice-stream2/commit/241680a479da497890f5dcc24221b17ec2f65356

mccollam, 2016-04-16:
I started hacking on this over here: https://github.com/mccollam/etcher

This adds a usage display (not super useful yet as there are no real options yet) and adds functionality to allow the user to specify an image to use on the command line.  If one is specified (and exists, etc.) it will be automatically filled in when the UI loaded.  This is really nice, as it means if you specify an image and have only one valid device to write to, everything just ticks along and is ready to start as soon as Etcher loads.

I didn't create a pull request as there are a few things I think might be iffy.  It adds a new dependency (on minimist, used for argument parsing) and is my first foray into node.js/angular/electron, so there's almost certainly a much nicer way to pass in the image name than what I am doing.  But if it's of any use or interest I'm happy to do whatever cleanup/changes you like. :)

alexandrosm, 2016-04-16:
@mccollam great to see you here :) hopefully you'll have a lot more time to
hack on resin projects soon :)

jviotti, 2016-04-18:
@mccollam Thank you very much for the PR! I'll continue discussion there :)