balena-io/etcher #334: Operation not permitted, write

inspector71, 2016-04-17:
Received this error, simply clicked Need Help, went back to Etcher, clicked the burn button again, and it started writing the image.

Previously, wrote exactly the same image with Win32DiskImager no problem.

At end of write, Etcher seems to have failed to auto-unmount, despite reporting to the contrary.

jviotti, 2016-04-18:
Hey there @inspector71 ,

Thanks for reaching out! The write permission issue you're having there most probably comes from Windows policies with regarding to direct I/O to physical drives (see https://github.com/resin-io/etcher/issues/271#issuecomment-208129752 for more info), which has been biting us a lot since the beginnings of Etcher.

Some questions to aid debugging:

- What version of Windows are you running?
- Can you actively reproduce the issue (maybe it happens once every three or four times you burn)?
- Does it happen usually at the beginning of the write? (maybe within the first 15%).
- What is the state of your drive *before* you start the writing? (does it have a mountable file system?)
- Does the issue disappear if you completely erase your drive (so nothing can be mounted) before starting the write?

inspector71, 2016-04-18:
> physical drives (see #271 (comment)

1. Windows 7 64bit
2. ... pending testing
3. The very start before anything seems to be written
4. I think the card was unformatted
5. ... pending testing

gillou1664, 2016-04-18:
I just had the same problem (it happens during the first 5%, on win 10).
I solved it by running the app as an administrator.

inspector71, 2016-04-18:
FWIW, the account I was using is an Administrator account.

gillou1664, 2016-04-18:
I may be wrong but I think that in Win 7, even if you are administrator, you don't have the privileges to do anything and you can need some rights elevation. 
Try launching the app with right-click + "Execute as administrator"

jviotti, 2016-04-19:
Hi @gillou1664 

When you start the application, the elevation dialog will prompt you to re-run the application as administrator, so there should be no difference is manually doing it yourself.

This permissions issue should be due to Windows policy when writing directly to physical drives, and not whether *your account* has permissions to perform certain operations (e.g: shouldn't matter if you're an admin).

Can you actively reproduce if you don't run as administrator?

inspector71, 2016-04-19:
On 19/04/2016 10:28 PM, "Juan Cruz Viotti" <notifications@github.com> wrote:
>
> Hi @gillou1664
>
> When you start the application, the elevation dialog will prompt you to
re-run the application as administrator, so there should be no difference
is manually doing it yourself.
>
> This permissions issue should be due to Windows policy when writing
directly to physical drives, and not whether your account has permissions
to perform certain operations (e.g: shouldn't matter if you're an admin).

Those are my suspicions as well which is why I prefaced my note about
running as an Admin with 'FWIW'. Didn't think it was relevant but just in
case.

Another note in the same vein could be UAC. Is it possible UAC settings are
involved? I doubt it but just throwing that out there.

Also, I wonder if any other apps based on Electron have had any similar
issues? Might be an avenue worth looking down.

I can do some testing soon but not immediately.

sergiopjf, 2016-04-20:
Same error (EPERM: operation not permited, write)

Every time you click the flash button

Tested in Windows 7 64bit and Windows 10 64bit, with flash drive and SD card.

Some times, it displays the error while formatting the flash drive as Windows Explorer asks you to format the device after error.

Sorry my bad english.

jviotti, 2016-04-20:
Thanks for all your help guys. The fact that Windows provides little information about the reason of the error, and the fact that only a few users can reproduce (and not me sadly) makes it very difficult to find the root cause of the issue, but we'll eventually get there. Any information/observation would be really valuable!

> Some times, it displays the error while formatting the flash drive as Windows Explorer asks you to format the device after error.

This is very nice information. Let's see if I can try to reproduce.

jviotti, 2016-04-20:
By the way, may I ask what image are you trying to write with Etcher? Maybe I can manage to reproduce that way.

inspector71, 2016-04-20:
I wonder if you're on to something ... could be a 7z format problem?

http://dietpi.com/downloads/images/DietPi_RPi-(Jessie).7z
>

jviotti, 2016-04-20:
@inspector71 Are you burning the 7z directly, without uncompressing first?

inspector71, 2016-04-20:
Indeed I guess I am, how silly I feel!

jviotti, 2016-04-20:
:) I doubt its the cause of the issue anyway, but I'll give it a go. We'll be supporting writing compressed images very soon: https://github.com/resin-io/etcher/issues/325

sergiopjf, 2016-04-20:
Any image.

Examples: 
ubuntu-14.04.4-desktop-i386.iso
lubuntu-14.04-desktop-i386.iso
linuxmint-17-cinnamon-64bit-v2.iso
Windows8.1_x64.iso
Windows10_x64.iso
Hiren's.BootCD.15.2.iso

jviotti, 2016-04-20:
@inspector71 I couldn't reproduce with the `7z` image, but noticed I was getting an `EINVAL` error due to  your DietPi image being unaligned. I'm in the process of fixing the issue (aligning the streams ourselves) in https://github.com/resin-io/etcher/issues/348

jviotti, 2016-04-20:
@sergiopjf @inspector71 I spend most of my day debugging this without too much luck. I was able to reproduce the error only once, after trying like a hundred times with all sorts of images in Windows 10 x86_64.

What I did notice, is that if another process tries to scan the the available drives while Etcher is burning, Etcher will complain with the EPERM error mentioned here, since it loses exclusive access (which is silly, if Etcher has exclusive access, the other process should be denied, not Etcher).

That would be caused by a myriad of things. Are you running any anti-virus, HD optimisers, etc that could be conflicting with Etcher? Can you try disabling those and see if it makes any difference?

I also noticed that sometimes scanning the available drives form inside Etcher takes a several seconds (usually when the OS slows down a bit), which can cause the EPERM issue since it keeps scanning the drives once more after the write has started.

sergiopjf, 2016-04-20:
I I disabled the antivirus, but continues with the error. No HD optimisers in machine.

No problema in scanning the available drives here.

sergiopjf, 2016-04-20:
Another detail, after the error the device is no longer listed in the usb safe removely in tray icon.

inspector71, 2016-04-21:
@jviotti Thanks for spending so much effort on this project, FOSS in general. Brilliant.

I'm running MSE but actually had realtime protection disabled at the time of testing. 

I've been able to write an .img file from that DietPi image BTW, after extracting from the .7z image, whether aligned or not. FWIW, aligning within compression is beyond my understanding at presence. FWIW I used Bandizip to extract the image. However I'm fairly certain Bandizip wraps a 7z.exe file for 7z support so I may have actually been using 7z in effect, for all I know.

The other program I'm running, that may cause similar behaviour to what you're suggesting, would be my file manager of choice ATM, Q-Dir.

Best of luck, keep up the great work. I'll help where I can. I think Etcher is a good idea. I'm a bit worried about the Electron choice but if it can be made to work, great. Not sure there's any particular cross-platform framework that would guarantee to be a better choice. That said, I'm not a desktop developer,

Not sure if it would make a difference but don't forget, I'm on Windows 7 64 bit, not Windows 10.

I think it's great that Etcher will have a goal of allowing drag drop and support for compressed images.

jviotti, 2016-04-21:
Thanks a lot for your collaboration and support, its very appreciated :)

> Another detail, after the error the device is no longer listed in the usb safe removely hardware in tray icon.

@sergiopjf Etcher has a feature to auto-unmount devices on completion, to avoid having users to go through the "Safely remove" step and simply remove the drive instead. This logic should only happen then a write succeeded (independently on validation results). Did it happen even when the write was interrupted with our infamous `EPERM`?

> Best of luck, keep up the great work. I'll help where I can. I think Etcher is a good idea. I'm a bit worried about the Electron choice but if it can be made to work, great. Not sure there's any [...]

Thank you very much :) The main reason behind our technology choice for this project is to take a challenge on pushing the web to the edge with something that wasn't done before. Interacting with physical drives is very straightforward in GNU/Linux and OS X (we didn't have any reported issue on those OSes). Windows, on the other side, is a completely different world. Native applications mitigate these kinds of issues because they use the Windows C/C++ APIs, exposed by the operating system, which seem to handle all these edge cases, and do all sorts of incantations for everything to work fine. Sadly, Windows doesn't provide much information in this topic and the code is closed-source, so we have to find our way by ourselves.

> Not sure if it would make a difference but don't forget, I'm on Windows 7 64 bit, not Windows 10.

Yeah, I know, but don't have access to a Windows 7 machine, everyone seems to have upgraded already, and VMs lead to all sort of false positives with Etcher. Maybe I can install it?

> I think it's great that Etcher will have a goal of allowing drag drop and support for compressed images.

Regarding the drag and drop, it has been implemented, but someone reported it not working correctly on Windows anymore for some reason. Probably a regression or Electron bug, but didn't have the time to investigate yet.

***

As a side note, there is something that should be worth testing, which is trying to reproduce the issue using the [Etcher CLI version](https://github.com/resin-io/etcher-cli). In order to run this, you need to install [NodeJS](https://nodejs.org/en/). Once that's installed, you can run the following commands in `cmd.exe`:

```sh
> npm install -g etcher-cli
> etcher path\to\image.img
```

The tool will interactively prompt you to select from your connected drives, and will perform a write.

inspector71, 2016-04-21:
On Thu, Apr 21, 2016 at 11:14 PM, Juan Cruz Viotti <notifications@github.com
> wrote:

> Thanks a lot for your collaboration and support, its very appreciated :)
>
No worries. I find it incredible how many people do free work for FOSS
without any expectation of credit. There should be website called
opensourceheroes.com

> Best of luck, keep up the great work. I'll help where I can. I think
> Etcher is a good idea. I'm a bit worried about the Electron choice but if
> it can be made to work, great. Not sure there's any [...]
That's very impressive to read. Hopefully the challenge with prove
achievable.

> Interacting with physical drives is very straightforward in GNU/Linux and
> OS X (we didn't have any reported issue on those OSes). Windows, on the
> other side, is a completely different world.
>
Completely? Windows has had Posix support for over a decade or more. Is
that something you can look at?

> Native applications mitigate these kinds of issues because they use the
> Windows C/C++ APIs, exposed by the operating system, which seem to handle
> all these edge cases, and do all sorts of [...]
I'm not an expert but the model you are following isn't brand new. It's
been well trodden by Mozilla (Firefox, Thunderbird, etc) and presumably
Google (Chromium), although yeah writing to hardware devices may not be
something they've explicitly done. That said, I presum they've manipulated
hardware devices with the recent 'HTML5' API explosion aimed at getting the
web to be more useful on native devices. Loosely off the top of my head,
there's recent APIs implemented for Bluetooth, webcams (microphones ?) for
WebRTC and mouse pointers at least. Then there was the recent File API
implementation. I remember reading a few years ago about that. Internally
in Gecko, file access across the various native platforms required an
overhaul in terms of how it was handled behind the scenes if not when
exposed to content. Fairly sure they wrote new code or at least cleaner,
more consistent wrappers around existing code. If you want I could try and
look up some references. I don't want to send you on a wild goose chase
overall but both Gecko and Chromium are substantial projects and have open
source code you may be able to peruse for some tips or perspective.

> Not sure if it would make a difference but don't forget, I'm on Windows 7
> 64 bit, not Windows 10.
>
> Yeah, I know, but don't have access to a Windows 7 machine, everyone seems
> to have upgraded already, and VMs lead to all sort of false positives with
> Etcher. Maybe I can install it?
>
Nuts. Not having a test suite is a PITA. VMs not behaving as expected is
alas predictable but annoying. I won't be upgrading to 10 though at work am
forced to use it in the immediate term on a new machine until I get time to
nuke it. I hate 10.

> I think it's great that Etcher will have a goal of allowing drag drop and
> support for compressed images.
>
> Regarding the drag and drop, it has been implemented, but someone reported
> it not working [...]
Fair enough.

> ------------------------------
>
> As a side note, there is something that should be worth testing, which is
> trying to reproduce the issue using the Etcher CLI version
> <https://github.com/resin-io/etcher-cli>. In order to run this, you need
> to install NodeJS <https://nodejs.org/en/>. Once that's installed, you
> can run the following commands in cmd.exe:
>
> > npm install -g etcher-cli> etcher path\to\image.img
>
> The tool will interactively prompt you to select from your connected
> drives, and will perform a write.
>
Can do :)

jviotti, 2016-04-21:
> I presum they've manipulated
hardware devices with the recent 'HTML5' API explosion aimed at getting the
web to be more useful on native devices. Loosely off the top of my head,
there's recent APIs implemented for Bluetooth, webcams (microphones ?) for
WebRTC and mouse pointers at least. Then there was the recent File API
implementation.

Right, but browser rendering engines implement this features natively, and then expose an API to the JavaScript world, so they have access to the operating system APIs. They don't do the whole thing in JavaScript, which is what we're trying to achieve.

> Not having a test suite is a PITA. 

This repository mostly contains tests for the frontend of the application. The module that performs the actual write is [resin-image-write](https://github.com/resin-io-modules/resin-image-write), which has an extensive test suite that performs on the file system, by writing a file to another file (since devices are represented by files), however there are certain things that can't be covered by an automated testing suite very easily, like writing to real physical drives.

inspector71, 2016-04-21:
Nuts, my quoting got all butchered by gmail or github in my previous comment.

I wasn't sure you were that glued to doing everything in JS. 

Yeah things will always boil down to testing on real devices.

inspector71, 2016-04-21:
```
C:\Users\****\Downloads\DietPi_RPi-(Jessie)(1)>etcher  "DietPi_v111_RPi-(Jessie).img"
? Select drive \\.\PHYSICALDRIVE1 (7.9 GB) - PciE SDHC-Card
? This will erase the selected drive. Are you sure? Yes
Burning [                        ] 0% eta 757m15s
You don't have enough privileges to run this operation.
Run a new Command Prompt as administrator and try running this command again.

If this is not the case, and you're trying to burn an SDCard, check that the write lock is not set.
```

inspector71, 2016-04-21:
Win+R
cmd
cmd.exe now has administrator privileges.

```
C:\Users\pd\Downloads\DietPi_RPi-(Jessie)(1)>etcher "DietPi_v111_RPi-(Jessie).img"
? Select drive \\.\PHYSICALDRIVE1 (7.9 GB) - PciE SDHC-Card
? This will erase the selected drive. Are you sure? Yes
Burning [========================] 100% eta 0s
Your flash is complete!
```

jviotti, 2016-04-21:
@inspector71 Thanks a lot for the testing. I forgot mentioning that the Etcher CLI doesn't handle elevation for you, yet.

The good news is that running from the CLI seems to work, and the EPERM issue is not thrown at all. They both share the exact same routine to perform the writing which points out to my theory of the drive scanner interval not having time to stop before the flash starts.

I pushed a branch that contains a commit to add a big delay (8s) between clicking "Flash" and actually starting the write, which should validate the theory.

Instructions to test:

```sh
> git clone https://github.com/resin-io/etcher
> cd etcher
> git checkout fix/334/eperm-windows
> npm install
> bower install
> npm start
```

inspector71, 2016-04-21:
Sounds like progress. My git/Github Desktop installation on this Win7 machine isn't 100% ATM so I can't do that CLI update and test just yet. Is there a download/unzip/then npm install/test option that's easy enough? If not, I'm getting git/Github Desktop cleaned up now.

jviotti, 2016-04-21:
@inspector71 I'll build it for you and upload the package here in a second.

jviotti, 2016-04-21:
@inspector71 There you go, sorry for the delay, I'm on a very slow internet connection at the moment: https://www.dropbox.com/s/87frwh8d6ln1ckf/Etcher-win32-x64.zip?dl=0

inspector71, 2016-04-21:
Not a problem. Gave me time to get my GitHub Desktop setup almost up to
speed again :)

Sorry to hear about your slow speed. Sucks.

inspector71, 2016-04-21:
Wellll...

That didn't go so well.

Buggered if I know how but that version has, believe it or not, bricked my
laptop.

jviotti, 2016-04-21:
@inspector71 Ugh, that sucks. I've just retried the application I sent you here and it seems to work fine. Could it be something else? Did the app run?

inspector71, 2016-04-21:
I've a bricked laptop to deal with. I bet Etcher wiped / imaged the wrong
drive.

The CLI version doesn't distinguish between fixed or portable media.
Guessing the flipping GUI version doesn't either and somehow picked my hard
drive.

This is more than 'sucks'. This is my primary machine that's probably just
been nuked.

My help stops here. Etcher us alpha ware I'm afraid, not version 1.

Maybe in a month when I *might* have my setup somewhere near normal again
...

jviotti, 2016-04-21:
@inspector71 Sorry to hear that. Both the CLI and the GUI re-use the same logic with regarding to hiding system drives for you, and there hasn't been any updates on that side for a while, so if Etcher displayed your hard-drive, it should have been displayed it all this time. 

Let me know if there is something I can help with.

sergiopjf, 2016-04-21:
My test with the CLI. View the screenshot:
[image]

closed 2016-06-12 by jviotti