balena-io/etcher #336: Implement auto-update functionality

jviotti, 2016-04-18:


probonopd, 2016-04-19:
Check [AppImageUpdate](https://github.com/probonopd/AppImageKit/blob/master/AppImageUpdate.AppDir/README.md) - it can do binary delta updates which means that only the changed parts of the binary need to be downloaded during an update.

jviotti, 2016-04-19:
Hi @probonopd ,

Thanks a lot for the suggestion. 

> We are using the ISO 9660 Volume Descriptor #1 field "Application Used" to store this information.

Do you have more documentation, or maybe an example, on how to add the update information this way?

***

What's the recommended way to ask end users to perform the upgrade? I wouldn't want to ask them to download an application `AppImageUpdate.AppImage` to perform this task. Maybe I can embed distribute this command line tool along with the `AppImage`, but renamed to something like `EtcherUpdate`?

probonopd, 2016-04-19:
> Do you have more documentation, or maybe an example, on how to add the update information this way?

Yes, [here](https://github.com/probonopd/AppImageKit/blob/master/AppImageUpdate.AppDir/usr/bin/appimageupdate#L121-L130).

>  Maybe I can embed distribute this command line tool along with the AppImage, but renamed to something like EtcherUpdate?

Absolutely, consider the `appimageupdate` script and the AppImageUpdate.AppImage GUI around it a sample implementation. You could instead have your app perform the same kind of thing natively. You could do a GUI inside your app similar to https://sparkle-project.org/.

jsejcksn, 2016-05-18:
Is this the last issue before 1.0 is released? I was looking around the site and the issues and didn't see any indicators about release date or progress toward 1.0. Thanks!

jviotti, 2016-05-18:
Hi @jsejcksn ,

The only things delaying us for v1.0 are currently:

- Auto updating functionality in all operating systems.
- Only elevating the process that performs the flashing, which I'm working on at the moment.

jsejcksn, 2016-05-18:
@jviotti Thanks for the quick reply. Is there a page other than [milestones](https://github.com/resin-io/etcher/milestones) where you post info about release progress? A twitter or blog, perhaps?

jviotti, 2016-05-18:
Hi @jsejcksn ,

Not sadly, I'm afraid, but its definitely something we should be dealing with better. As far as I've seen, most OSS projects make use of Milestones for that, so probably cleaning it up a bit should work.

Do you have another examples/suggestions?

jsejcksn, 2016-05-18:
No, I think Milestones is the simplest solution since you're already using GitHub. Maybe just check to see that all the blocking issues are actually listed in the relevant milestone.

And if want to increase awareness at any level, you could add a link to the [Milestones sorted by closest due date](https://github.com/resin-io/etcher/milestones?direction=asc&sort=due_date&state=open) (in case you begin to use that feature) in the [readme](https://github.com/resin-io/etcher/blob/master/README.md) and on [Etcher.io](https://etcher.io).

jviotti, 2016-05-18:
Sounds good, thank you very much for the suggestions. I'm making a note to cleanup the milestones and link appropriately in the README and in the website.

jsejcksn, 2016-05-18:
Also, a quick topic detour: you might consider enabling https on Etcher.io (you can get a cert for free from Let's Encrypt, but I'm sure Resin already has one and the Etcher.io domain could be added to it for re-use).

probonopd, 2016-05-18:
@jviotti 

> Do you have more documentation, or maybe an example, on how to add the update information this way?

First, you would have to inject the update information into the AppImage, second you would have to upload the AppImage and the AppImage.zsync to an URL that matches the URL specified in the update information.

**Step 1: Inject the update information into the AppImage**

Assuming you have a file `Etcher-linux-x64.AppImage` in your current working directory, then the following command would tell the updater to look for `Etcher-linux-x64.AppImage.zsync` at the URL `https://resin-production-downloads.s3.amazonaws.com/etcher/latest/Etcher-linux-x64.AppImage.zsync`:

```
echo "zsync|https://resin-production-downloads.s3.amazonaws.com/etcher/latest/Etcher-linux-x64.AppImage.zsync" | dd of="Etcher-linux-x64.AppImage" bs=1 seek=33651 count=512 conv=notrunc 2>/dev/null
```

**Step 2: Upload the AppImage and the AppImage.zsync to an URL that matches the URL specified in the update information**

You would have to copy both `Etcher-linux-x64.AppImage.zsync` and `Etcher-linux-x64.AppImage.zsync` to `https://resin-production-downloads.s3.amazonaws.com/etcher/latest/` for this to work.

When you have a new version, simply put the updated  `Etcher-linux-x64.AppImage.zsync` and `Etcher-linux-x64.AppImage.zsync` at the *same* location.

At this point, AppImageUpdate should work. AppImageUpdate is really just an example GUI around the (work-in-progress but useable) [appimageupdate](https://github.com/probonopd/AppImageKit/blob/master/AppImageUpdate.AppDir/usr/bin/appimageupdate) bash script. You could put `appimageupdate` (and its dependency [zsync_curl](https://github.com/probonopd/zsync-curl)) inside `usr/bin` of the AppImage, and call `appimageupdate` from within your app and even have a little GUI around it; or reimplement it in the language of your app.

jviotti, 2016-05-19:
Hi @jsejcksn ,

Yeah, we should definitely do that. I wonder if GitHub pages allows us to configure an SSL certificate. /cc @craig-mulligan.

jviotti, 2016-05-19:
Hi @probonopd ,

Thank you for the extensive explanation on this. I'll get on it shortly!

jsejcksn, 2016-05-20:
@jviotti I just came across this project management tool that integrates with GitHub Issues: [Waffle.io](https://waffle.io). You might be interested to see if it can fit with your development progress in regard to what we discussed earlier.

In regard to SSL: [It seems that](https://developer.ubuntu.com/en/blog/2016/02/17/how-host-your-static-site-https-github-pages-and-cloudflare/) you need to be using [CloudFlare](https://www.cloudflare.com/) if you want to host on GitHub pages with a custom domain using https.

jviotti, 2016-05-20:
Hi @jsejcksn ,

Funnily enough, we're already using Waffle.io for Etcher (https://waffle.io/resin-io/etcher), but at a very basic level. Could be a good way forward, I'll definitely keep it in mind!

flokX, 2019-03-03:
Any new progress here? It's very annoying to check every time the changelog and then download the latest version.
Maybe it would be also a pleasure if the version information would also show if the installed version is outdated.

incyi, 2020-09-22:
4 years to update ...

librafrog, 2021-03-02:
Should note that we still download manually after checking the changelog. There is a option to auto update, maybe it's doing something in the background and its core. 👎🏻

jviotti, 2022-08-15:
Closing stale issues

closed 2022-08-15 by jviotti