balena-io/etcher #355: Package and distribute the Etcher CLI

jviotti, 2016-04-22:


jviotti, 2016-11-01:
As @lurch suggested, the ideas mentioned in http://stackoverflow.com/a/14601786/1641422 might be worth trying.

jviotti, 2017-02-27:
@jhermsmeier Did an amazing work discovering some nodejs command line options to reduce the memory footprint in the CLI, which we should include by default: https://gist.github.com/jhermsmeier/c21270571f5ab2a0dd1262521fee647f

```
node --max-executable-size=8 --max-old-space-size=16 --max-semi-space-size=1
```

jhermsmeier, 2017-02-27:
These arguments would need to be passed in the `execArgv` option to [`child_process.fork()`](https://nodejs.org/api/child_process.html#child_process_child_process_fork_modulepath_args_options) to have an effect:

```js
var child = childProcess.fork( './worker', argv, {
  execArgv: [
    // max size of the old space (in MB)
    '--max-old-space-size=16',
    // max size of a semi-space (in MB), the new space consists of two semi-spaces
    '--max-semi-space-size=1',
    // max size of executable memory (in MB)
    '--max-executable-size=8',
  ],
})
```

Note that forcing garbage collection (scavenge) with `--expose-gc` and calling `gc(true)` every n-th chunk during streaming to collect excess `Buffer`s appears to be more effective in keeping memory consumption down (see [block-write-stream/index.js#L29-L37](https://github.com/jhermsmeier/node-block-write-stream/blob/43517d20846110e8390471f5b4a904ed4d25d245/index.js#L29-L37) and [block-write-stream/index.js#L63-L72](https://github.com/jhermsmeier/node-block-write-stream/blob/43517d20846110e8390471f5b4a904ed4d25d245/index.js#L63-L72)) than setting these limits.

jviotti, 2017-05-11:
I can confirm this issue was sorted out completely by https://github.com/resin-io/etcher/pull/1409

closed 2017-05-11 by jviotti