balena-io/etcher #357: Hash Check on Image

DexterInd, 2016-04-24:
It would be great to have an Md5 or other hash check on an image.  They're always large, and they can be corrupted in download.

jviotti, 2016-04-24:
Hi @DexterInd ,

Definitely something we want to implement in Etcher soon, thanks for raising it up.

I wonder where should Etcher attempt to read the checksum from. Maybe we can have some sort of convention where Etcher attempts to read a plain text file with a `md5` extension that contains the same file name of the image, placed in the same directory of the image?

Needs much more thinking.

DexterInd, 2016-05-13:
Sorry I missed the response on this!  I'm not sure if we even need to automate it.  We currently recommend our PC folks to use win32diskimager, and it has a simple checkbox (not very nice looking, but works) that calculates the checksum.  When someone has trouble with imaging, one of the first things we ask is to check the checksum.  We package a text file and png file with every image we compress, so it needs to be checked manually.

Just a thought.  These are usually large images, good to check it sometimes.

jviotti, 2016-05-13:
Hi @DexterInd ,

I see, that makes a lot of sense, and its trivial to implement. What type of checksum do you use? Etcher already computes a CRC32 checksum of the image as part of the verification process, we could just show that on the "Finish" screen.

What do you think?

DexterInd, 2016-05-16:
You could definitely show that.  We use an MD5 checksum already with win32diskimager, but either one should work I think.

jviotti, 2016-05-16:
Hi @DexterInd ,

Awesome, I'll ping our designer for input on this and will implement shortly!

jviotti, 2016-05-18:
[image]

closed 2016-05-19 by jviotti

jviotti, 2016-06-12:
Hey there @DexterInd ,

I noticed you still recommend Win32 Disk Imager for Windows in your instructions. Is there something else missing from Etcher that's keeping you from recommending it for Windows as well?

shawaj, 2016-08-20:
Does your software do an md5 hash check or some other?

jviotti, 2016-08-21:
Hi @shawaj,

We do calculate a CRC32 checksum for the image, which is displayed once
the flash completes, as shown in the screenshot above in the thread.

shawaj, 2016-08-21:
Any chance you can do an MD5 hash instead/as well?

This is more common and is what is shown for win32 disk imager and other
tools so will be an easier way for people to transition from one to the
other.

jviotti, 2016-08-22:
Hi @shawaj ,

Yeah, lets do it. I agree on showing both. I've opened an issue here: https://github.com/resin-io/etcher/issues/643

shawaj, 2016-08-22:
Awesome, thanks a lot!

We'll be switching over all our tutorials to etcher as soon as that's done.

One other request would be the ability to create images :-)

jviotti, 2016-08-23:
Sounds exciting! I'll make sure to ping you when this is available.

> One other request would be the ability to create images :-)

What do you mean by this?

shawaj, 2016-08-23:
As in, win32 disk imager can take an SD card and make a .IMG file from it.

Would be good to do that with yours too!

Cheers

jviotti, 2016-08-23:
I see. I believe https://github.com/resin-io/etcher/issues/266 is the one covering that, so keep an eye on that one as well :)

shawaj, 2016-08-23:
Thanks so much. Keep up the great work!!

Do you know of a utility on windows to do a hash/checksum of an existing
file?

Can your software do that too? Very useful for Pi images to be able to do
that

jviotti, 2016-08-24:
Hey there,

> Do you know of a utility on windows to do a hash/checksum of an existing
> file?

I recommend installing something like Cygwin, or MinGW, to access UNIX
utilities such as `md5` and `crcr32`.

> Can your software do that too? Very useful for Pi images to be able to do
> that

Etcher displays the final checksum of what was written to the drive on
the completion screen. Regarding hashing a specific image itself, I
believe that's out of the scope of Etcher's, and should be done by the
image publisher rather than by the user.