balena-io/etcher #387: Building etcher

stephdgenius, 2016-05-01:
How to build etcher on windows ?

jviotti, 2016-05-02:
Hi @stephdgenius ,

We have a [PUBLISHING guide over here](https://github.com/resin-io/etcher/blob/master/PUBLISHING.md) that showcases how to create distributable Etcher packages, but notice we build Etcher on UNIX based operating systems, so not sure if'd work at all.

Some things you have to consider:

- The build process is described in a `Makefile`. I believe something like Cygwin or MinGW that provides `make.exe` should work.

- The build process makes use of code signing. You can sign using your own certificates, or otherwise comment out [this line](https://github.com/resin-io/etcher/blob/master/Makefile#L94) and [this one](https://github.com/resin-io/etcher/blob/master/Makefile#L115).

Let us know how it goes!

jviotti, 2016-05-10:
Hey there,

I just merged https://github.com/resin-io/etcher/pull/395 which implements the build system for Windows as a `.bat` script you could run yourself. Check the PUBLISHING for more information.

closed 2016-05-10 by jviotti