balena-io/etcher #389: Interferes with some keyboard shortcuts

danhood, 2016-05-02:
Etcher seems to have some odd behavior regarding the `Cmd-Alt-I` shortcut on Mac OS X...

1. Etcher seems to steel this keyboard shortcut from all open apps - regardless of focus (if it wants to do something with that shortcut, it should only do so when it has focus)
2. I suspect bringing up the DOM inspector atop the GUI is not the desired behavior

jviotti, 2016-05-03:
Hey there @danhood 

> Etcher seems to steel this keyboard shortcut from all open apps - regardless of focus (if it wants to do something with that shortcut, it should only do so when it has focus)

Ugh, check this from the documentation (http://electron.atom.io/docs/v0.37.8/api/global-shortcut/):

> Note: The shortcut is global; it will work even if the app does not have the keyboard focus. You should not use this module until the ready event of the app module is emitted.

That makes no sense at all :/ We provide a way to open DevTools for debugging purposes, since it might be quite tricky for us to get to the bottom of an issue without looking under the hood a bit. Maybe there is a way we can provide this without keyboard shortcuts (menu entry, or something like that)?

> I suspect bringing up the DOM inspector atop the GUI is not the desired behaviour

Yeah, its quite ugly, and you need to re-position DevTools as a separate window. Sadly, I don't think there is a way to control how DevTools is opened. I'd love to be proved wrong.

jviotti, 2016-05-03:
Just an idea for the first question: We can unregister and register the keyboard shortcut by listening to the application focus event. Not very nice, but could work.

closed 2016-05-24 by jviotti