balena-io/etcher #410: Etcher on OS X El Capitan stalls at 99% with some zip files.

jafine, 2016-05-14:
Etcher v1.0.0-beta.5 stalls at 99% writing from zip files that aren't compatible with unzip.

When trying to open said files with unzip e.g. the 2016 Dexter Industries Raspbian for robots image, I see:
```
76 extra bytes at beginning or within zipfile
  (attempting to process anyway)
error [2016.05.09_Dexter_Industries_jessie.zip]:  reported length of central directory is
  -76 bytes too long (Atari STZip zipfile?  J.H.Holm ZIPSPLIT 1.1
  zipfile?).  Compensating...
   skipping: 2016.05.09_Dexter_Industries_jessie.img  need PK compat. v4.5 (can do v2.1)

note:  didn't find end-of-central-dir signature at end of central dir.
  (please check that you have transferred or created the zipfile in the
  appropriate BINARY mode and that you have compiled UnZip properly)
```
If I unzip to .img with 7z from the cli, then Etcher completes the write without any issues.

jviotti, 2016-05-16:
Hi @jafine ,

Weird issue! Looks like your zip file contains wrong size information on the headers, and thus confuses Etcher. 

I believe we should be able to compensate this by ignoring the zip header size if the input stream doesn't send any more data. Would you mind attaching the Zip that you're using here so I can test and implement?

jafine, 2016-05-16:
The zip file I was using is available here: http://sourceforge.net/projects/dexterindustriesraspbianflavor/

This might be relevant http://unix.stackexchange.com/questions/115825/extra-bytes-error-when-unzipping-a-file

jviotti, 2016-05-16:
Cheers, I'll be looking at this shortly.

jviotti, 2016-07-25:
I've dug into this, and it seems to be a bug in [unzip2](https://github.com/glebdmitriew/node-unzip-2), the Zip library we're using. The process hangs for this particular zip, and starts eating a lot of RAM and CPU, until it eventually dies. 

I didn't dive into the actual implementation issue, but seems like [yauzl](https://github.com/thejoshwolfe/yauzl), another Zip module, doesn't have this issue, plus it uncompresses faster, and with less RAM usage. We'll switch to this new module, and see how things go.

jviotti, 2016-07-25:
`yauzl` indeed fixes the issue. I'll send a PR to upgrade `etcher-image-stream`, and this problem should be fixed in the next Etcher version.

closed 2016-07-25 by jviotti