balena-io/etcher #412: Application doesn't have permission or ask for password to use application on default Linux Xubuntu 14.04

gameslayer, 2016-05-16:
This isn't a big deal for me but by default this application cannot work unless you open the terminal and either run it with sudo -s or gksu being the best easiest quick fix to ask for the a password to run the software, maybe it should have this added into the shortcut it makes depending on the distro so it does "Works for everyone, no more complicated install instructions."

leillo1975, 2016-05-16:
Have the same problem in Ubuntu 16.04 64Bits. I made a workaround:
1. Install "Main Menu" (alacarte) 
2. Search Etcher in the menu elements
3. Edit Etcher in properties and write "gksudo" before command
4. Now when a select Etcher in Dash (or menu), it prompt me to introduce my sudo password

jviotti, 2016-05-16:
Hey there,

Thanks for the suggestions. We're working on solving the root cause of this problem, which is only elevating the forked process that actually performs the writing. I'm locally baking a commit to implement this (I have it working already, but needs some polishing).

I'll hopefully release it in the later version, stay tuned!

danrabbit, 2016-05-16:
Wanted to make a note that GKSU is deprecated and should not be used as per OP's recommendation. The proper thing to do would be to use [policykit](https://wiki.gnome.org/Projects/PolicyKit)

jviotti, 2016-05-17:
Hi @danrabbit,  

We'll go with `pkexec` as per your suggestion!

gameslayer, 2016-05-17:
so what would be the multi distro solution to this to elevation issue

jviotti, 2016-05-17:
Hi @gameslayer ,

Instead of elevation the complete application at the beginning, which doesn't play nice when executing the app from the Unity launcher for example, we'll only elevate a forked process that performs the writing, for which we can show nice GUI dialogs for all distributions based on `pkexec`.

closed 2016-05-26 by jviotti

gameslayer, 2016-05-26:
@jviotti Sounds great hope it works well multi platform :)

jviotti, 2016-05-26:
@gameslayer I've did some extensive tests on the Linux machines I could access too and it works fine as long as the user has either `gksudo`, `kdesudo` and `pkexec`. We'll see! I'm publishing a new version in a couple of hours, don't hesitate to ping me if you find an edge case!

jviotti, 2016-05-26:
Hi @gameslayer ,

Time to give it a go: http://www.etcher.io :D

danrabbit, 2016-05-26:
I can confirm that the latest version of etcher asks for permission on elementary OS. Good work guys! Might wanna drop the "sudo" line from the directions on the website ;)

jviotti, 2016-05-26:
Sweet @danrabbit , thanks for the testing, I'll drop the website instructions in a minute :)

gameslayer, 2016-06-01:
Problem is solved with newer version tested on Xubuntu.

PR #423 (2016-06-21): [WIP] Only perform elevation of writer process
This is a big PR that solves a hard issue we've been carrying over for
too long. The Etcher GUI application now runs an intermediary script
called `writer-proxy.js` which elevates itself if needed, spawns the
Etcher CLI piping `stdout` to a temporary file, tails that file and uses
IPC communication to share back the results with the Etcher GUI.

Fixes: #412
Fixes: #351
Fixes: #287
Fixes: #444