balena-io/etcher #413: Cannot chose writable space for Live Ubuntu USB

timtjtim, 2016-05-16:
I can't remember which app does it (perhaps `usb-creator-gtk`) but I remember being able to choose the free space for writing files when booting a Live USB such as Ubuntu.

Is this supported / could this be supported?

jviotti, 2016-05-16:
Hi @timtjtim ,

I don't think I understand the feature request very well. What do you mean by writable space? USB drives written by Etcher shouldn't be read-only, so you could mount it back and put any other files you want.

miniShrooms, 2016-05-19:
Hi @jviotti,

I believe @timtjtim is speaking about the option for persistent storage when creating a live USB for Linux OS's like Ubuntu, Mint, and more. What it does is allow the live USB to have files written and stored on it (like a normal OS) and actually allow for the files and changes to to be saved when closing, disconnecting, and rebooting back into the live USB. Additionally, **UNetbootin** is the application that has this feature; and honestly, I'm a big fan of it myself. I'd really like to see a feature like this implemented into Etcher if possible.

timtjtim, 2016-05-19:
@jviotti Yes, @miniShrooms is correct. That's the idea.

jviotti, 2016-05-22:
Hey there,

Thanks for the very detailed explanation. I'll play around with `UNetbootin` and try to research how this is implemented, and if it'd be reasonable to include it in Etcher.

m4gn3to, 2016-06-16:
I'm too interested in persistent storage, so I can install some packages in my usb only once, and also only have to disable the screensaver once 👍 
Thanks

Shou, 2016-11-25:
Reading UNetbootin's source, the approach it takes is to [create a `casper-rw` file](https://github.com/unetbootin/unetbootin/blob/787323b2a4595c50a400f1024ca141e4708c7b7d/src/unetbootin/unetbootin.cpp#L4259) in the root directory by padding it with `\NUL` to the user specified size—limited to 4GB by FAT32—and then running `mke2fs` on said file. On Windows, it [ships with a `mke2fs.exe` executable](https://github.com/unetbootin/unetbootin/blob/787323b2a4595c50a400f1024ca141e4708c7b7d/src/unetbootin/unetbootin.cpp#L4266). Another approach is to make an EXT2 partition labelled `casper-rw` which doesn't have the 4GB limitation. In order to facilitate persistence it also [seems to enable a kernel option for persistence](https://github.com/unetbootin/unetbootin/blob/787323b2a4595c50a400f1024ca141e4708c7b7d/src/unetbootin/unetbootin.cpp#L3626) which it then [adds to the GRUB config](https://github.com/unetbootin/unetbootin/blob/787323b2a4595c50a400f1024ca141e4708c7b7d/src/unetbootin/unetbootin.cpp#L3679).

alexandrosm, 2016-11-26:
Another modification of the same kind is to extend the last partition to
the end of the card. Raspbian and resinOS both need/do that. It sounds like
we should curate a list of "post-write interventions" people may want to
make to their images, and then see how we can encode these in the etcher
format. Once we have that in place, we should find how to open these up to
vanilla images. We could, potentially, offer the users manual ways of
triggering them from the success screen, or maybe having some kind of
heuristic?

lurch, 2016-11-28:
>  It sounds like we should curate a list of "post-write interventions" people may want to make to their images, and then see how we can encode these in the etcher format.

:+1:  I was thinking along exactly the same lines after @petrosagg 's [LKL](https://github.com/lkl/linux) talk.

Nokia808, 2017-05-02:
Hi. I'm not sure whether the following could assists you about adding this feature or not. Please look for this program:

https://www.pendrivelinux.com/

it is for Windows only. But look to it, my be of benefit.

Look to last change log for Universal USB Installer - Easy as 1 2 3:
https://www.pendrivelinux.com/universal-usb-installer-easy-as-1-2-3/

it say: Universal-USB-Installer-1.9.7.7.exe – March 21, 2017 – Changes
Added casper-rw creation progress banner. Corrected casper slider max size relative to fat32 vs ntfs selection. Add Windows to Go VHD option. Added more info for syslinux warning. Add NTFS format option. Add filesystem type to drivelist. Include message box to notify of NTFS filesystem requirement for Windows to Go option. Add larger than 4GB casper-rw support for Ubuntu if filesystem is NTFS (creates 4th partition table) – Note removes EFI boot option. Correct broken Ubuntu and Linux Mint installer options when no format option is chosen.

Also, there is other program:
https://www.linuxliveusb.com/

But 1st program (Universal USB Installer - Easy as 1 2 3)

Best.

jo-chemla, 2017-07-19:
Hi, is this feature still planned on the todo list ? Would be a great addition to etcher, for example compared to Rufus.

lurch, 2017-07-19:
@jviotti Perhaps this is a feature that might fit in with the "plugin" system you've been thinking about?

jviotti, 2017-07-19:
@jhermsmeier Could this be done generically? As far as I understand we simply have to update the partition table and mark certain partitions as read-write, right? If so, how do we know which? We now have a lot of tooling to work with partition tables, so maybe we're half way there for this already.

If we need image specific information to do this properly, then I guess this will fit inside the catalog idea.

lurch, 2017-07-19:
I once looked at this many moons ago (before even working on the Etcher project, so my info is a bit foggy and out of date) and IIRC it involves casper filesystems, syslinux, various other bits and bobs... ;-)

Don't forget that even when enlarging a partition, you have to update both the MBR (to modify the raw partition size) **and** the filesystem itself (to update the size of the filesystem inside the partition).

casearis, 2018-02-09:
A lot of distro's are available with supported persistence built in, puppy linux, porteus, slackware, slitaz, and yes also Ubuntu and Debian. Through past experience Puppy has it made, runs in ram (very fast), secure, comes in different flavours.  The most common reasons  users desire persistence, are Desktop settings, User installed packages and personal files. Most systems either use "casper" or "cow" for writing changes (although limited). Is easy achieved as shown in multibootusb, Which, with the size of portable storage devices nowadays, is another option which i personally would like to see. As achieved with yumi for windoze and the afore mentioned multibootusb.

pythlang, 2018-02-13:
+1 bump for persistence capability. it's basically a commodity these days for Mac users to be able to do anything like this

ethanent, 2018-03-18:
+1 from me as well. This is really important for Mac users who are dual-booting Linux.

inferno986return, 2018-03-26:
+1 Definitely would like the feature so I can try out distros and use them for a few things.

aadilayub, 2018-08-09:
This feature would be very useful for my friend who wants to try Linux on her Mac.

Mattybootsy, 2018-08-31:
so will this be added? I recently made an Ubuntu drive but I cant save my files

casearis, 2018-09-04:
Not sure if persistence will be added to etch..  for ubuntu distro's,  unetbootin and liveusb cover that.. however, needs to be fat32/16 filesystem. Try multiboot-usb, good for testing live iso's. Can also handle persistence for deb/ubu , also supports ext4 fs. 

Most live iso's have persistence built in. But that's another story...

alichtman, 2019-02-09:
This would be an awesome feature. Etcher is second to none in terms of ease of use, but not being able to enable persistence makes it unusable for most of my purposes.

ohbus, 2019-08-02:
I have been researching during my free time to contribute and make this feature available in balena, but my point is I have figured this out for Debian/Ubuntu Systems already but not sure how to implement it for other Distros.

Would love to have some help.

alichtman, 2019-08-04:
@ohbus Can you describe your solution for Debian/Ubuntu?

lurch, 2019-08-05:
>  I have figured this out for Debian/Ubuntu Systems already but not sure how to implement it for other Distros.

@ohbus I'd assume that https://unetbootin.github.io/#distros and https://github.com/unetbootin/unetbootin/wiki/howitworks and https://github.com/unetbootin/unetbootin/wiki/howitworks would be good starting points?

ohbus, 2019-08-05:
Yes

> @ohbus I'd assume that https://unetbootin.github.io/#distros and https://github.com/unetbootin/unetbootin/wiki/howitworks and https://github.com/unetbootin/unetbootin/wiki/howitworks would be good [...]

Yes you are absolutely right.

They are the starting point, but need to implement it for different OS and archtype.  

I'm a little busy right now, will surely get back to this once I get some time.

Thank you.

PastLoves, 2019-09-02:
Ohbus, is there a way to make it with ParrotOS. If you can make a secondary application that has that option. It may greatly enhance your solutions because errors would pop up.

ulidtko, 2019-12-27:
Ah yes. **Quarter Gig** of an app to do what a **0.000075 Gig** `dd` tool does — but with eyecandy. 

What a waste of my SSD capacity!

The app *so beautifully* displays the 16 GiB flash capacity I picked to boot from — only to put a measly 1.3G of casper-rw free space on it.

What a waste of my flash capacity!

Honestly guys? I'm disappointed. **Four orders of magnitude bloat** is not okay. My SSD space is not that cheap as you'd like to think. My time is not that cheap, to start over with a tool that doesn't suck, and can produce a usb stick which can boot *and* store another ISO image on it.

Keep not caring.

Forage, 2019-12-28:
> Honestly guys? I'm disappointed. **Four orders of magnitude bloat** is not okay. My SSD space is not that cheap as you'd like to think. My time is not that cheap, to start over with a tool that [...]

This is just an utterly useless and highly disrespectful post.
We are here to suggest and discuss improvements to make the application better, not to cry and bitch about something that had not even been implemented yet.

You sound as if you are being forced to use this application that has been made available to all of us for free.
Go make an effort to look for an alternative that actually promises to do what you want. It is not until you actually put some time, effort and/or money into it and it does not live up to its promises that you have some form of right to complain about it.

timtjtim, 2019-12-28:
> Ah yes. Quarter Gig of an app to do what a 0.000075 Gig dd tool does — but with eyecandy.

Use dd then.

ehaliewicz, 2021-11-17:
> This is just an utterly useless and highly disrespectful post. We are here to suggest and discuss improvements to make the application better, not to cry and bitch about something that had not even [...]

Honestly, how is it disrespectful?  This tool uses literally over a hundred times of every user's space than necessary.  That's what I would call disrespectful.

ehaliewicz, 2021-11-17:
> This is just an utterly useless and highly disrespectful post. We are here to suggest and discuss improvements to make the application better, not to cry and bitch about something that had not even [...]

You don't need to write an alternative to dd to know that a 300MB alternative to dd is extremely wasteful.

ulidtko, 2021-11-18:
... I apologize as well, my original rant has obscured the real point I was making:

 * the program puts **just 1.3G of casper-rw persistent space** — on a 16G flash stick.
   As a result, the live USB cannot even fit another ~1.5G ISO image on it — leaving 10+ GiB of space unused. 

That should make it clear why I posted that in this issue specifically.

Gipper925, 2022-02-22:
Rufus supports persistence.

pythlang, 2022-02-22:
> Rufus supports persistence.

We know this already. Rufus doesn’t run on Unix platforms

casearis, 2022-02-22:
"gufw", supports persistence.    Do usb stuff

casearis, 2022-02-22:
Apologies   " dusb  "do usb stuff

casearis, 2022-03-05:
Sorry was away from home when i commented, the software i was referring to is "mkusb". A GUI for linux - supports persistence.  Have only used etcher for raspberry pi images, I do however pre format SD to FAT32 before i use any writing applications.

jsmdy, 2023-02-03:
was expecting this functionality on etcher also :(

nileshtrivedi, 2024-03-29:
This feature would be great.

It's 2024 and I can't believe I am unable to set up a bootable linux USB thumbdrive with persistence without already having a Linux or Windows machine around.

-     Etcher runs on Mac OS but does not provide persistent volume feature
-     Rufus has persistent volume feature but does not run on Mac OS.
-     Unetbootin supports persistent volumes and runs on mac OS but has not been updated since 2021 and therefore does not offer modern versions of Linux desktop OS
-     Fedora Media Writer runs on Mac OS but does not support persistent volume feature.
-     `livecd-iso-to-disk` supports persistent volume but does not run on mac OS.