balena-io/etcher #414: with: support creating a multi-boot USB drive

markstos, 2016-05-16:
This may be outside of the scope of the project but having a single USB stick that contains multiple bootable images inside it would be very useful.

Some other projects do this, but none seem to excel at it.

jviotti, 2016-05-17:
Hi @markstos ,

Thanks for the suggestion. It might be out of the scope of the core functionality provided by Etcher, but we're planning to implement a plugin framework, so users can build and distribute plugins to do all sorts of crazy workflows, like cloning drives, streaming the image file from a URL, etc; so this might be a great use case as well.

The plugin framework work has not been started yet, and it'll probably take a while to mature, but just to collect as much information as possible on this issue for when that is ready:

- Do you know about a couple of applications that performs this, so we can try?
- Do you happen to know how this works internally?

markstos, 2016-05-17:
Here are some references for multiboot USB on Linux:

   * https://wiki.archlinux.org/index.php/Multiboot_USB_drive
   * http://multibootusb.org/
   *
http://www.pendrivelinux.com/multiboot-create-a-multiboot-usb-from-linux/

Most recently, I had trouble getting "MultiBootUSB" to work for me. I think
there may have been a problem relating to BIOS vs UEFI booting. I never
quite figured out why it wasn't working my case. But I've got this working
in the past, and *love* having a single USB "rescue" stick that can hold
several different Linux rescue "disks" on it was well as several different
installation ISOs.

andreasmalling, 2016-09-07:
There is also a great Windows example of this called YUMI which makes it possible to have a stick with both multiple Linux Distros and Windows images.
http://www.pendrivelinux.com/yumi-multiboot-usb-creator/

Forage, 2016-09-27:
My choice goes to YUMI at the moment as well. It's got a crude UI and I'm forced to resort to a Windows VirtualBox to use it, but it gets the job done. Multiboot as well as persistent storage (#413) for that matter.
Multiboot and Linux support with an efficient UI would make me switch to Etcher right away.

dragon788, 2017-04-15:
Easy2Boot is another good example of multi booting USBs, but it is more of an ISO loading framework than an actual "flasher".

Catley94, 2018-11-07:
Any idea why everytime I use Etcher to burn an ISO to a USB that it won't make it bootable? For example if I want to burn the Windows 10 ISO onto a USB, it doesn't make it bootable, or Ubuntu etc. I've tried this with the absolute latest version of Etcher and on Windows 10/Ubuntu 18.04.

Thanks,

Sam

Forage, 2018-11-07:
@Catley94 How is your issue related to this issue? Please create a separate issue for, well, different issues.

tinohager, 2022-08-25:
We have tried with [Ventoy](https://github.com/ventoy/Ventoy) to create a stick works so far well the only problem is that we can not simply download the finished image via the Etcher or the portal.

closed 2025-05-12 by markstos

Forage, 2025-05-12:
@tredondo More settings does not make it a better program, I'd say it's the opposite.
As for the file size, Etcher is an Electron app, as such it includes the Chromium browser engine.
If you consider UNetBootin to be better, why complain here and not just use the other program?

tredondo, 2025-05-12:
@markstos I was also considered Ventoy, until I read about the [closed-source blobs](https://github.com/ventoy/Ventoy/issues/2795), [on which the author never commented](https://github.com/ventoy/Ventoy/issues/132). I'd rather not risk running a [backdoored program](https://www.reddit.com/r/linux/comments/1buhnrs/is_ventoy_safe_in_light_of_xzliblzma_scare/?rdt=46090).

@Forage: I apologize. I was just puzzled about the lack of seemingly a critical feature after so many years and though I was missing something, but it turns out it's just hard to implement.

markstos, 2025-05-12:
@tredondo The Ventoy author did comment on the blobs, five days ago. You can read the response here: https://github.com/ventoy/Ventoy/issues/3224