balena-io/etcher #415: Add as snap package in Ubuntu

ericoporto, 2016-05-16:
Just a thought. It's ridiculously easy to pack.

jviotti, 2016-05-17:
This is actually something we've been thinking about, so thanks for the reminder. I've been skimming through the Ubuntu Developer Portal documentation, and will be looking forward to give it a go soon.

jviotti, 2016-09-06:
I'm closing this since after some internal discussion, we decided that a Snappy package is not beneficial for us over AppImages, at least for the time being.

closed 2016-09-06 by jviotti

sidahmed-malaoui, 2017-05-22:
It would much easier to install if you add it as a snap package.

Jacalz, 2018-02-19:
Yeah I would love to get etcher as a snap image since it would update it self and work across most Linux distros 👍

Feichtmeier, 2018-03-21:
I would also love to install it as a snap package. Many great companies have joined the snap band waggon. Including Microsoft with skype and vscode, spotify and intellij with all their IDEs.
Snaps are greatly included in the gnome-software app which comes preinstalled on many distros. They even promote snaps which are new so people become interested and install it.

DarkLite1, 2018-05-01:
+1 for snap package

NatoBoram, 2018-05-21:
> We decided that a Snappy package is not beneficial for us over AppImages
> https://github.com/resin-io/etcher/issues/415#issuecomment-244827814

Can we have more details on that?

MeRuslan, 2018-08-29:
+1 for snap

jviotti, 2018-09-20:
Sounds interesting! I remember I hit various issues when trying to build a Snap for Etcher some time ago, but we can definitely revisit it. @thundron is currently the maintainer of Etcher, so he can take this forward. 

@thundron Looks like @Mikerr1111 has an Etcher fork with some initial setup, so maybe you can coordinate the contribution?

aurelijusrozenas, 2018-10-09:
So, maybe reopen issue? ;)

armintelker, 2018-10-22:
Please reopen this. 
Your app as a snap package would make my/our life so much easier. 
I am switching pretty often between multiple linux distributions for testing. Snap makes the switch easier and my install script is much more readable since it is not depending anymore on so much different package managers. Your app is one of my last apps that is not published via snap. 

As a small motivation I am sponsoring a sixpack of beer, coke or a beverage of choice for successfully publishing a snap package into the snap store.

HattinGokbori87, 2019-02-22:
I use Etcher everyday for my work (Thanks!!). I also like to see it as a snap if doing so doesn't increase pressure  on devs. It's not a top priority as Etcher is already available at deb.etcher.io but many end-users prefer snaps for different reasons (auto updates, delta updates, not having to add another repository which slows down update checking etc).

Jacalz, 2019-02-22:
> I use Etcher everyday for my work (Thanks!!). I also like to see it as a snap if doing so doesn't increase pressure on devs. It's not a top priority as Etcher is already available at deb.etcher.io [...]

Not to mention the fact that far from everybody uses Ubuntu or Debian. We are a lot of users that have snap enabled didstros like Solus, Manjaro and so on. It’s not just about easy updates. A well set up snap is also in theory more secure than a regular old package file...

Jacalz, 2019-02-22:
If snap isn’t your thing, go with Flatpak on Flathub instead. It has pretty much all the good points of snap while having better desktop integration, it is available on more distorts and it is fully open source compared to snap. Probably a better alternative to be quite honest 👍

lurch, 2019-02-22:
Obligatory XKCD link: https://xkcd.com/927/
:wink:

the-spyke, 2020-04-07:
Please reconsider adding a snap package.

> Snappy package is not beneficial for us over AppImages

Snap gives users at least easy updates and free discoverability for you in the Snap Store.

lurch, 2020-04-07:
#2019 (and https://github.com/flathub/flathub/pull/1363 ) says that it was impossible to get Etcher packaged as a "flatpak" because of the low-level permissions Etcher requires in order to be able to write to disks at the block-level. I wonder if the same thing might affect "snap" packaging? :man_shrugging:

the-spyke, 2020-04-07:
@lurch It may be impossible yet, because of sandbox-first approach chosen for snap/flatpack. But you need to try `classic` confinement with snap first and open a ticket on Snapcraft if it isn't enough, so snap developers will be aware of the issue and make a plug/portal/solution for such applications.

lurch, 2020-04-07:
@the-spyke I don't work on Etcher, I'm just a helpful 3rd-party user :slightly_smiling_face:

thundron, 2020-04-07:
We looked into it and for now we haven't currently got any plans to bring a snap/flathub package.
If anyone feels like giving it a try and also managing/supporting it, we're always open to contributions! :D

the-spyke, 2020-04-14:
@thundron Raspberry Pi image flasher works fine as a snap: https://github.com/popey/imager-snap

xlmnxp, 2020-06-14:
@lurch there isn't any problem to package it as Snap package
see "Raspberry Pi Imager" available at Snapcraft
https://snapcraft.io/rpi-imager ([source code for YAML](https://github.com/popey/imager-snap))

enoch85, 2022-07-13:
I'm surprised that you decided to not use snaps. AppImage is legacy by now.