balena-io/etcher #416: Custom window border style for Ubuntu

Corsseir, 2016-05-16:
I will be more than grateful, if you could add custom window border styl for Ubuntu version, which would be compatible with Ubuntu default style. Mockup:

[image]

jviotti, 2016-05-17:
Hi @Corsseir ,

Thank you very much for the suggestion. I've looked at this, but sadly this doesn't seem to be supported by [Electron](https://github.com/electron/electron), our cross-platform runtime (only in OS X for now).

I encourage you to open an issue on the Electron repo for them to implement this for GNU/Linux and link back to this one, so I can stay in the loop. Happy to provide this styling once its supported in Electron!

jviotti, 2016-06-22:
Hi @Corsseir ,

The only way I can think of having this on GNU/Linux would be to create a frameless window and implement the close/minimise buttons ourselves which sounds like a bad idea.

I'll close this issue since we're already specifying the `titleBarStyle: 'hidden-inset'` option in `BrowserWindow`, so if Electron decides to add support for GNU/Linux, the feature will land automatically.

closed 2016-06-22 by jviotti