balena-io/etcher #420: Error: Error executing 'etcher.wrapper'; return code: -1

kasalolek, 2016-05-20:
ubuntu,

jviotti, 2016-05-20:
Hey there,

Thanks for reaching out. Is that all you get? What Ubuntu version are you running?

`etcher.wrapper` is just this script: https://github.com/resin-io/etcher/blob/master/scripts/desktopintegration, maybe running it can help us debug further?

kasalolek, 2016-05-23:
Thats all I get.

Distributor ID: Ubuntu
Description: Ubuntu 16.04 LTS
Release: 16.04
Codename: xenial
hycom@LAP00186:~$

Mate Desktop
java 1.6u45

jviotti, 2016-05-23:
That's too bad. Can you try manually executing the AppImage with the commands described in [this wiki page](https://github.com/probonopd/AppImageKit/wiki/FUSE#fallback)?

```sh
sudo mount -o loop Some.AppImage /mnt
/mnt/AppRun
```

jviotti, 2016-05-24:
Also, seems that appImages display more debugging information if you open them with the `DEBUG=1` environment variable set.

jviotti, 2016-06-22:
I'm closing this issue since we recently moved to only elevating the writer process (https://github.com/resin-io/etcher/pull/423) instead of the whole application, which should solve this bug as a consequence. Please re-open if the issue happens again after the new version is released.

closed 2016-06-22 by jviotti