balena-io/etcher #421: Disallow selecting System Hard Drive

volumio, 2016-05-20:
Especially for novices, it would be safer to disallow to write on non-removable drives. I tried on win10 x64 and I could select my hard drives ...

jviotti, 2016-05-20:
Hi @volumio ,

Thanks for reaching out. Etcher is supposed to filter out non-removable drives from the drive selection list, if that was not the case for you, then it probably means that there is an edge case we didn't consider.

The actual module that performs the drive filtering is [drivelist](https://github.com/resin-io-modules/drivelist). It'd be amazing if you can run this `.bat` script (https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat) in your PC and paste back what you get.

volumio, 2016-05-20:
Here you are:

device: "\\\\.\\PHYSICALDRIVE1"
description: "ST1000LM024 HN-M101MBB"
size: 1000202273280
mountpoint: "D:"
name: "D:"
system: False

device: "\\\\.\\PHYSICALDRIVE1"
description: "ST1000LM024 HN-M101MBB"
size: 1000202273280
mountpoint: "E:"
name: "E:"
system: False

device: "\\\\.\\PHYSICALDRIVE0"
description: "Samsung SSD 850 EVO 250GB"
size: 250056737280
mountpoint: "C:"
name: "C:"
system: True

I assume you take the system value to determine if an hard drive is removable or not. Probably in this case, given that I have multiple HDs it fails to reckon it. 
I am looking at drivelist to see if I can help...

PS: Kudos to the whole team for this, I was sharing the same feeling of shame for my users writing SD Cards... Thanks!

volumio, 2016-05-20:
Side note. A script you could launch on win could be:

wmic logicaldisk get caption,description,filesystem

No removable drive
Caption  Description          FileSystem
C:       Disco rigido locale  NTFS
D:       Disco rigido locale  NTFS
E:       Disco rigido locale
Z:       Connessione di rete  FAT32

Removable drive

Caption  Description          FileSystem
C:       Disco rigido locale  NTFS
D:       Disco rigido locale  NTFS
E:       Disco rigido locale
F:       Disco rimovibile     FAT
Z:       Connessione di rete  FAT32

The issue is: output is localized, here you'll see in italian... But we can filter the results to show only removable drives: 

wmic logicaldisk where drivetype=2 get deviceid, volumename, description

And I get: 

Description       DeviceID  VolumeName
Disco rimovibile  F:        boot

Which is easily parsable...

jviotti, 2016-05-20:
Hi @volumio ,

Thanks for the quick response and the suggestions. This project was initially running `wmic` to get the output, but I remember we got stuck with at some point (can't remember exactly), and thus moved to calling `wmi` programatically in VBScript. Looks like at the moment, the module is only marking the drive where Windows its installed (e.g: `C:` in your case) as a system drive.

I believe we can make use of your `drivetype=2` suggestion programatically, but I'll have to look at it (the docs are quite terrible). 

I'll keep you updated!

volumio, 2016-05-20:
My 2 cents:
what if you just use node.execSync and parse the output? No VB gimmicks and should be done... If I have some spare time I'll try it...

PS: Do you guys agree if I for this and use it as ImageWriter for Volumio (of course keeping resin credits!)?

jviotti, 2016-05-20:
Hi @volumio ,

> what if you just use node.execSync and parse the output? No VB gimmicks and should be done... If I have some spare time I'll try it...

Yeah, that was the original approach. You can see the older code by going back to the beginning of the project with git. The problem was:

- It was hard to maintain, since custom parsing needed to be done for each OS (right now each script outputs the same format).
- There was something we needed to do that couldn't be done with `wmic` directly (some sort of edge case, or check, again I don't really remember), so we switched to VBScript to interact with WMI directly (what `wmic` uses in the background as well).

> PS: Do you guys agree if I for this and use it as ImageWriter for Volumio (of course keeping resin credits!)?

Definitely, go ahead, we encourage users to fork and customise to their needs (e.g: artwork, or anything else). Currently, we're the only ones doing actual release builds from Etcher, so don't hesitate in pinging me if you have any issues with the build system.

volumio, 2016-05-20:
Thanks! I'll let you know also if I find a way to integrate a fix in wmic!

jviotti, 2016-05-24:
Hi @volumio ,

I've sent a PR (https://github.com/resin-io-modules/drivelist/pull/76) which I believe fixes the issue. Do you mind giving a go to see if it works for you?

volumio, 2016-05-24:
Nope, all the drives are reported as system drive (my card reader as well)...

> win32.bat
device: "\\\\.\\PHYSICALDRIVE1"
description: "ST1000LM024 HN-M101MBB"
size: 1000202273280
mountpoint: "D:"
name: "D:"
system: True

device: "\\\\.\\PHYSICALDRIVE1"
description: "ST1000LM024 HN-M101MBB"
size: 1000202273280
mountpoint: "E:"
name: "E:"
system: True

device: "\\\\.\\PHYSICALDRIVE0"
description: "Samsung SSD 850 EVO 250GB"
size: 250056737280
mountpoint: "C:"
name: "C:"
system: True

device: "\\\\.\\PHYSICALDRIVE2"
description: "TS-RDF5 SD  Transcend USB Device"
size: 7945620480
mountpoint: "F:"
name: "F:"
system: True

jviotti, 2016-05-24:
Hi @volumio ,

Thank you for the quick answer. Do you mind trying again? I pushed some changes to the PR to include some debugging information as well: https://github.com/resin-io-modules/drivelist/blob/dd4660bd60c45974e9ba552439ea2f949f980720/scripts/win32.bat

volumio, 2016-05-24:
Here it is 

device: "\\\\.\\PHYSICALDRIVE1"
description: "ST1000LM024 HN-M101MBB"
size: 1000202273280
mountpoint: "D:"
name: "D:"
# type: "Fixed hard disk media"
system: True

device: "\\\\.\\PHYSICALDRIVE1"
description: "ST1000LM024 HN-M101MBB"
size: 1000202273280
mountpoint: "E:"
name: "E:"
# type: "Fixed hard disk media"
system: True

device: "\\\\.\\PHYSICALDRIVE0"
description: "Samsung SSD 850 EVO 250GB"
size: 250056737280
mountpoint: "C:"
name: "C:"
# type: "Fixed hard disk media"
system: True

device: "\\\\.\\PHYSICALDRIVE2"
description: "TS-RDF5 SD  Transcend USB Device"
size: 7945620480
mountpoint: "F:"
name: "F:"
# type: "Removable Media"
system: True

jviotti, 2016-05-24:
Hi @volumio ,

Thank you very much again! This leads us to the right way. For some reason, Windows documentation says the drive type would be `Removable media other than floppy` rather than `Removable Media` :)

I'll update the PR and we should be good to go. I'll see if I can check that the type *starts* with "Removable" instead, just to be on the safe side, since maybe the message changes between Windows versions, and I see that the `M` in "Media" is capitalised in your output, while its lowercase on mine.

jviotti, 2016-05-24:
I've updated the script: https://github.com/resin-io-modules/drivelist/blob/4919cf471b41a288d15772cf7da76fcfd66708c8/scripts/win32.bat, it should work fine now.

jviotti, 2016-05-25:
I got my hands on a PC exhibiting a very similar behaviour and my PR fixed the issue. Please re-open if it persists for you!

closed 2016-05-25 by jviotti