balena-io/etcher #430: Drop images on window navigates to it

amka, 2016-05-23:
If drop image onto Etcher window then window will navigate and load this image instead of Etcher html.

Could be fixed with this:

```javascript
document.addEventListener('dragover',function(event){
  event.preventDefault();
  return false;
},false);

document.addEventListener('drop',function(event){
  event.preventDefault();
  return false;
},false);
```

[image]

jviotti, 2016-05-24:
Hi @amka ,

Thats very funny, I'd never have though of dragging an actual image to the window :) Your snippet looks good, however consider that we support drag and drop on the first step, so that code might get in between that feature.

I believe Electron allows us to inspect the file name on certain drag events, so we could deny the drag depending on the extension.

Happy to take a PR if you feel like working on it, otherwise I can handle it myself.

amka, 2016-05-24:
Hi @jviotti, I'm just writing app with drag-n-drop and made some tests with Etcher. 
I'm novice in JS and don't think can make PR in reasonable period.

jviotti, 2016-05-24:
Hi @amka ,

No worries, I'll take over on it, thank you very much for the heads up! I'm actually thinking we could PR this fix straight to Electron.

closed 2016-06-08 by jviotti