balena-io/etcher #431: Can activate "Flash!" button with keyboard while a write is in process

mccollam, 2016-05-23:
While Etcher is performing a write, it is still possible to activate the "Flash!" button by using the spacebar while the Etcher window has focus.  This is handled gracefully by popping up a message about a write being already in progress, so it's a minor cosmetic issue.  But it seems like it would still be nice to just prevent this behavior.

Steps to reproduce:
1. Launch Etcher.
2. Begin an image write.
3. Press the spacebar.
4. Observe that an error box occurs.

jviotti, 2016-05-24:
Hi @mccollam ,

Awesome observation! I can't reproduce in OS X though, so it might be a Linux thing. I'll give it a go there.

jviotti, 2016-05-24:
Hi @mccollam ,

Turns out I can't reproduce on Ubuntu 16.04 LTS neither :/ Could it be something specific to your setup? What distro, desktop environment, or window manager are you running?

mccollam, 2016-05-24:
Yes, I can do this by clicking "Flash!" and then hitting the spacebar a few seconds later.  Screenshot of the result attached; I can make a video if that's more helpful.

Ubuntu 16.04 amd64 running stock Unity, Etcher 1.0.0-beta6

fader@lacquer:~$ uname -a
Linux lacquer 4.4.0-22-generic #40-Ubuntu SMP Thu May 12 22:03:46 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

[image]

jviotti, 2016-05-24:
Thanks for the explanation, I can reproduce now. I was attempting to press "space" once I lost focus on the button. Sending a fix right now, thanks a lot!

closed 2016-05-24 by jviotti