balena-io/etcher #439: Created desktop shortcut cannot launch application on Ubuntu

orange-buffalo, 2016-05-26:
System: Ubuntu Gnome 16.04.
Preconditions: Answer "Yes" when asked to create desktop shortcut during the first application launch.
Steps: 
1. Go to Activities and locate Etcher shortcut
2. Launch application using shortcut
3. 
ER: user is asked for admin password (like in GTK Greeter, for example)
AR: application crashes with "Please run as as root or administrator"

jviotti, 2016-05-26:
Hi @orange-buffalo,

This is indeed a very annoying issue, apologies for that! We already have a fix for it in the following pull request: https://github.com/resin-io/etcher/pull/423, although that one is currently blocked due to some Windows issues.

I'll split the commit and PR the Linux fix separately so we can have it as soon as possible. I'm planning to publish a new version today or tomorrow, so you'll probably have the fix very soon!

orange-buffalo, 2016-05-26:
Hi Juan,

That's awesome! Thanks for the great tool!

closed 2016-05-26 by jviotti

jviotti, 2016-05-26:
Hey there @orange-buffalo ,

I'm thrilled to announce that the new version, `v1.0.0-beta.7`, has been published and contains the fix to your issue. Please download from http://www.etcher.io. Thanks!

orange-buffalo, 2016-05-27:
Hi Juan,

Just checked it - works like a charm. Thank you!