balena-io/etcher #448: Etcher 1.0.0-beta.7 causes GNOME Shell to become unresponsive on Ubuntu 16.04

dlech, 2016-06-01:
I just tried installing Etcher on an Ubuntu 16.04 using the GNOME 3 desktop. When I get to the step of attaching the "drive" (microSD card in USB adapter in my case), the entire desktop becomes unresponsive for at least 10 or 20 seconds. It seems to become responsive again after I remove the SD card, but it could be just a coincidence that at the point I get too impatient and pull it is when whatever is locking things up times out.

Any tips on how to troubleshoot?

jviotti, 2016-06-06:
Hi @dlech ,

Hm, this sounds strange. I never heard about something like this, and I usually run Etcher on Fedora with GNOME 3 without any issues, so I'm not sure how to troubleshoot. Some thoughts:

- Can you reproduce this every single time?
- Does it happen only with a certain drive, or with every removable one?
- The module that performs the drive scanning is https://github.com/resin-io-modules/drivelist, which in GNU/Linux, calls the following bash script: https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh. Maybe you can try running that script directly and see if you can get it to block, outside Etcher itself?

I have an Ubuntu 16.04 box, so will try to install GNOME 3 on it and see if I can reproduce, in case its an Ubuntu quirk.

dlech, 2016-06-06:
> Can you reproduce this every single time?

Yes. I also timed it this time and it locks up for around 30 seconds. It does not matter if you leave the drive connected or remove it - it will stay locked up for the same amount of time.

> Does it happen only with a certain drive, or with every removable one?

I just tried a USB flash drive with the same results.

> Maybe you can try running that script directly and see if you can get it to block, outside Etcher itself?

I tried running this script and it works as expected (does not block).

jviotti, 2016-06-06:
This is very strange. I installed GNOME Shell on Ubuntu 16.04 but still couldn't reproduce. Just to make sure we're looking at the right thing, can you try executing Etcher on the same computer, but in a different desktop environment?

dlech, 2016-06-07:
I should know better than to not reboot before reporting a bug. After logging out and logging back in, it is working as expected. If it happens again, I will reopen th issue. Thanks for your time.

closed 2016-06-07 by dlech

jviotti, 2016-06-07:
Very funny! Glad you have Etcher up and running without any issues. Please don't hesitate in re-opening this issue if you encounter the same issue again.