balena-io/etcher #449: ~/.config/etcher is owned by root on Linux

dlech, 2016-06-02:
I have installed 1.0.0-beta.7 on Ubuntu 16.04. I noticed this because my backup program complained that it could not read `~/.config/etcher` in my home directory. On further inspection, I have found that the directory is owned by `root`.

    $ ls -l ~/.config
    ...
    drwx------  5 root  root   4096 Jun  1 09:58 etcher
    ...

Ideally, this should be owned by my user account. If it absolutely must be owned by root, it should be stored somewhere else outside of my home directory.

Related: #444

jviotti, 2016-06-06:
Hi @dlech ,

Thanks a lot for reporting this and sorry for the inconveniences. We have a PR in progress (https://github.com/resin-io/etcher/pull/423), which is mainly been blocked by some Windows issues coming from Electron itself to only elevate what's needed, however we have some major problems with AppImages, since forking a process with privileges inside an AppImage is not straightforward because FUSE security measures (see https://github.com/probonopd/AppImageKit/issues/146), but we'll push forward as most as we can for this; we're sure there is a solution to it!

jviotti, 2016-06-21:
I've just merged https://github.com/resin-io/etcher/pull/423, which introduces a big change to only elevate the writer process instead of the whole application, which should solve this issue.

closed 2016-06-21 by jviotti