balena-io/etcher #453: Install ISO in one Partition of usb Disk

sae13, 2016-06-07:
so i have two partition on my usb disk
first is sdc1 which i want it be empty and ntfs! cuz windows only can read first partition of usb disk

second is sdc2 which i wanna install my iso on that.

jviotti, 2016-06-07:
Hi @sae13 ,

The ISO already contains information about the partitions that should be created in your drive (a MBR for example). The partitions that are created in your drive are not up to Etcher, but up to your image, therefore we have to write the image to the beginning of your drive (otherwise it'll not boot).

sae13, 2016-06-08:
with gparted i make my usb disk two part. then I format part 2 as fat32 and make thats flag as bootable.
after that i install manjaro on thats partition on usb disk and it works with no problem

jviotti, 2016-06-09:
I see. I never did that before. I'll re-open for investigation, although I don't know how this would play with images that contain a non-trivial partition table. What software are you using to flash an image to a partition?

Donneur2lumiere, 2016-12-18:
lili usb creator do it without any problems. Perhaps only put iso in first partition sometimes. But it works

mymike00, 2017-06-17:
Yeah, LinuxLive USB Creator (Lili), unetbootin and another program doesn't touch the partition already present in the usb/hard disk. So it's possible to do, but instead of them I prefer Etcher because of the graphic, LiLi doesn't work on linux etc... I'll love you if you add a feature like that

mymike00, 2018-02-23:
@lurch yeah, they seems very similar

jhermsmeier, 2018-02-23:
Yeah, good catch @lurch!
Duplicate of #414

jhermsmeier, 2018-02-23:
Duplicate of #414

closed 2018-02-23 by jhermsmeier