balena-io/etcher #465: exFAT support

yhynerson1, 2016-06-09:
I just used this app for installing Windows 10 onto my Mac, sadly it doesn't support NTFS. Could you support formatting my flash drive in exFAT?

jviotti, 2016-06-09:
Hey there,

What do you mean by NTFS/exFAT support? Etcher doesn't deal with filesystems, it just blindly copies an image file contents byte by byte to the drive.

If you're looking to format your drive, then there are other programs available for such task.

yhynerson1, 2016-06-09:
No there aren't. Is this what a responsible person supposed to say?

jviotti, 2016-06-10:
I'm sorry but I don't really understand your comment.

> No there aren't

Are you talking about programs that can format to exFAT? If so, the default format tool for Windows supports exFAT, and Disk Utility from OS X seems to do as well.

yhynerson1, 2016-06-10:
I mean, when the app starts flashing the disk image, it reformats the disk and after flashing the image, the drive becomes read-only. Are there any ways to prevent that? Also, are there any options to make the  drive bootable?

jviotti, 2016-06-10:
@yhynerson1 Etcher doesn't format the drive, nor edit any drive attribute. It blindly copies bytes to it. The format the disk will have, and its attributes, like "bootable" or "read-only" are determined by the image, not by Etcher.

closed 2016-06-13 by jviotti